bamarni/pi64 #1: make a lite version

bamarni, 2017-03-14:
Similarly to Raspbian, there should be 2 releases : the lite one should contain a minimalist linux setup with basic networking, the second one would be full-featured (ssh enabled, avahi daemon etc.).

edegraaff, 2017-03-19:
When possible, add icsci support to, since more people use a nas at home, and... just booting from sd and then switch to iscsi could save a lot of problems with broken sd cards or usb drives. At this moment i have serious issues with the latest raspbian and iscsi drivers, try to get it working as a total, with connect directly during boot to the iscsi disk looks simple but it seems more complex ...

maxidev, 2017-03-20:
Hi there and thanks for your work, is there any issue or known problem with the latest build?
Regards

bamarni, 2017-03-22:
hi @maxidev, lately I was mostly focused on having a stable base system so I haven't tested much things, but there are for sure some issues.

Some might come from the 64-bit kernel which is not officialy supported at this point (cf. https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=158127), some others are missing from my end, for example the wlan interface is not enabled yet, I also haven't tested bluetooth.

commit c2cee56d (2017-05-10): make a desktop version (closes #1)
commit 79b34041 (2017-05-12): make a desktop version (closes #1)
commit 62635d04 (2017-05-12): make a desktop version (closes #1)