bamarni/pi64 #3: no gateway

kdg3737, 2017-03-30:
Hi

Thanks for making this distribution!

The problem I'm still having is that my rpi doesn't get his gateway information (DHCP - default settings). The IP and DNS settings don't seem to have any issues, 'route -n' doesn't show the gateway and executing 'sudo ip route add default via 192.168.0.1 dev eth0' temporarily solves the issue (although I sometimes have to execute it twice to have effect).

Any ideas what could be wrong and how to fix it?

kdg3737, 2017-03-31:
The problem I had is (probably) caused by our DHCP server, in case anyone else runs into this:

See RFC3442 - DHCP Client Behaviour
If the DHCP server returns both a Classless Static Routes option and
a Router option, the DHCP client MUST ignore the Router option.

So the Debian distribution (unlike many others) is doing what it's supposed to and seems very stable so far. 

Thanks again and sorry for the trouble.

bamarni, 2017-04-03:
Thanks for the follow-up on this, so closing for now :)

closed 2017-04-03 by bamarni