bamarni/pi64 #10: kernel headers

jedisct1, 2017-06-08:
Hi Bilal,

Is there an easy way to install the kernel headers?

I'd like to run Wireguard, but it comes with a kernel module, so it requires the presence of headers for the running kernel.

bamarni, 2017-06-23:
Indeed they are missing at the moment! Would it be enough if I run `make headers_install` as described [here](https://www.kernel.org/doc/Documentation/kbuild/headers_install.txt)?

I'll probably add a tarball release  just for kernel / modules / headers soon (to prepare #9).

jedisct1, 2017-06-23:
Hi Bilal,

Yes, that would do the trick!

bamarni, 2017-06-25:
In the meanwhile if you want to grab the tarball manually here it is :) (https://github.com/bamarni/pi64-kernel/releases)

clint3501, 2017-07-26:
does the tarball referenced above had the kernal headers?

Thanks

bamarni, 2017-07-26:
Yes they should.

bamarni, 2017-07-31:
fixed in latest release

JohnnyLeone, 2017-08-01:
Sorry for reopening this issue, but the kernel headers are referencing to old symlinks on the build machine. I cannot compile any kernel module because of that.

```
root@raspberrypi:~# ls -la /lib/modules/4.11.12-pi64+/
total 1800
drwxr-xr-x  3 root root   4096 Jul 30 18:52 .
drwxr-xr-x  3 root root   4096 Jul 30 18:50 ..
lrwxrwxrwx  1 root root     29 Jul 30 18:50 build -> /mnt/dev/pi64/build/linux-src
drwxr-xr-x 10 root root   4096 Jul 30 18:52 kernel
-rw-r--r--  1 root root 468314 Jul 30 18:52 modules.alias
-rw-r--r--  1 root root 480671 Jul 30 18:52 modules.alias.bin
-rw-r--r--  1 root root   4833 Jul 30 18:52 modules.builtin
-rw-r--r--  1 root root   6590 Jul 30 18:52 modules.builtin.bin
-rw-r--r--  1 root root 143089 Jul 30 18:52 modules.dep
-rw-r--r--  1 root root 208587 Jul 30 18:52 modules.dep.bin
-rw-r--r--  1 root root    302 Jul 30 18:52 modules.devname
-rw-r--r--  1 root root  58610 Jul 30 18:52 modules.order
-rw-r--r--  1 root root    327 Jul 30 18:52 modules.softdep
-rw-r--r--  1 root root 195103 Jul 30 18:52 modules.symbols
-rw-r--r--  1 root root 240025 Jul 30 18:52 modules.symbols.bin
lrwxrwxrwx  1 root root     29 Jul 30 18:52 source -> /mnt/dev/pi64/build/linux-src
```

bamarni, 2017-08-02:
@JohnnyLeone : kernel headers I've installed should be in `/usr/include`. I'm not familiar with installing a kernel module from source, but aren't the kernel headers enough? (cf. https://www.cyberciti.biz/tips/build-linux-kernel-module-against-installed-kernel-source-tree.html).

That linux sources folder it's referencing on the build machine is 1.6G of size, I'd really rather avoid shipping it unless absolutely necessary...

JohnnyLeone, 2017-08-03:
The headers from /usr/include are not the headers from the kernel tree. Without them we cannot build modules against the kernel. The size is huge, but we could shrink it if we import only the kernel architecture, module includes...

bamarni, 2017-08-13:
@JohnnyLeone : I'm currently looking at how to achieve that. This script seems relevant : https://github.com/torvalds/linux/blob/23ea3f62f3f90caaa7b4bf32690c341101228cca/scripts/package/builddeb#L318

It's the part which builds the kernel headers package for debian. I can probably extract what we need from there, I'll give it a shot soon.

jedisct1, 2017-08-24:
Nod :( Modules still cannot be built :(

bamarni, 2017-08-27:
I've just made a new kernel release containing stripped down kernel sources, using the same filters as the script linked above.

You can download it with `sudo pi64-update` or if you have a too old release directly from here : https://github.com/bamarni/pi64-kernel/releases

Could anyone confirm if modules can be built with this one?

jedisct1, 2017-08-27:
Yes!

It works! 👍

bamarni, 2017-08-28:
Awesome, thanks for checking

jedisct1, 2017-09-27:
Sorry for reopening this.

There is still a remaining issue.

The `scripts/basic/fixdep` file is an x86_64 executable, not an aarch64 one.

jedisct1, 2017-09-27:
Same for `bin2c`.

jedisct1, 2017-09-27:
Same for `recordmcount` :)_

jedisct1, 2017-09-27:
Same for `modpost`, etc....

drwatson32, 2017-11-12:
same here, file /usr/src/linux-headers-4.11.12-pi64+/scripts/basic/fixdep
/usr/src/linux-headers-4.11.12-pi64+/scripts/basic/fixdep: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=91599

mohakshah, 2017-12-01:
Same problem here. All the binaries in `/usr/src/linux-headers-4.11.12-pi64+/scripts/` are `x86-64`.