bamarni/pi64 #11: missing bcm2835-v4l2 kernel module

mash-graz, 2017-06-08:
i really like your pi64 images! 
the performance gain using ffmpeg/x264 in an amd64 environment is remarkable.

it would be very helpfull, if you could add/enable the bcm2835-v4l2 kernel module for raspberry camera support in your pi64 image releases as well.

see: https://blogs.s-osg.org/use-v4l2-cameras-raspberry-pi-3-upstream-kernel/

YuehChuan, 2017-06-08:
@mash-graz  Thanks, are there more details descriptions about  the performance improvement? and how you benchmark.
Like doing edge detection with openCV function the processing time from 10ms to 5ms.  Great to hear your good news :smile:

mash-graz, 2017-06-08:
in my case it's not so challenging. ;)
i just try to use a RP3 for video stream encoding and live streaming to an nginx-rtmp server over a very weak transmission line.
the RP3 h264 hardware encoder is nice, if you have enough bandwidth available, but it doesn't work very well, if you you need maximal image quality over limited network connections and content based variable bandwidth optimizations.
using the common raspian 32bit images, it's nearly impossible to handle this kind of compression by x264, but on arm64 you get about the doubled frame rates and it's becoming somehow acceptable at low resolutions.

YuehChuan, 2017-06-08:
Cool, so it mean 64bit register could handle more data at a time, that you got double framerate. 
Thanks for sharing, I would like to try, if there are some result will show here.

bamarni, 2017-06-23:
I'll look into it for the next release.

bamarni, 2017-07-11:
Just for the record, this commit suggests that it wouldn't work on arm64 (https://github.com/raspberrypi/linux/commit/97b35807cc4d3d0cf1d2c61f5ed0727551416ad0).

Nonetheless I'll try a build with that module just to see.

Truemedia, 2019-01-12:
Any updates on this, very keen to use the camera on 64 bit. 32 bit os has too many limitations for my uses.