bamarni/pi64 #12: upstream kernel support

mash-graz, 2017-06-08:
because the upstream kernel supports raspberry pi looks quite usable in the meanwhile, ready made images of pi64 utilizing this kind of kernels would be very useful.

see: 
https://michaelfranzl.com/2016/10/31/raspberry-pi-debian-stretch/
https://wiki.gentoo.org/wiki/Raspberry_Pi_3_64_bit_Install
https://wiki.debian.org/RaspberryPi3

bamarni, 2017-06-23:
What would be the advantage of this? The raspberry kernel is always rebased against the most recent kernel release (at the moment 4.11.6, https://github.com/raspberrypi/linux/commits/rpi-4.11.y?after=ccca5d14d5c584e80a00da948aaa6713cb6ffeff+174), however they keep dozens of patches on top of it for raspberry pi drivers, config, device tree etc.

I believe they're also trying to push their work to upstream but it can be quite a slow process. IMO their kernel is more reliable and up-to-date even though upstream is usable.

mash-graz, 2017-06-23:
the differences between upstream and the raspberry kernel are significant and the flow of patches and upstream integration doesn't work very well -- or as [eric anholt recently stated in his blog  about actual rapi3 related kernel improvements](https://anholt.github.io//twivc4/2017/06/19/twiv/): 

_"On the non-vc4 kernel side, I prepared and submitted pull requests for 4.13. We’ve got USB OTG support for the Pi Zero, SDHOST enabled by default finally, thermal enabled by default finally, and the RPi3 DT being installed for 32-bit kernels. Three of these **took months longer than they should have, because the Linux kernel has the worst software development process I’ve worked with since XFree86.**"_

right now most users will prefer the raspberry variant, because some devices are still not supported in upstream and its free drivers (e.g. analog audio output), but it's becoming better on every kernel release.

with the release of debian "stretch" arm64 is finally very well supported platform. even it's (upstream) kernels work out of the box and open a simple upgrade path (although they do not support all hardware features available in more recent kernel and its staging areas right now).

i therefore concentrated my recent activities more on this pure debian approach.

it's intriguing simple, to setup debian arm64 by `vmdebootstrap` (see: https://people.debian.org/~stapelberg/raspberrypi3/2017-03-22/). 
there are just some nasty bugs in the actual `vmdebootstrap` and `raspi3-firmware` packages, which have to fixed, but otherwise it works exceptional well.

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=845526
https://gitlab.com/mash_graz/raspi3-firmware/commits/fixes_by_mash

sure, your pi64 development is also a very nice and remarkable clean alternative. for those users, which are first of all interested in maximized raspian similarity and increased performance, pi64 is indeed a very well choice. but for those others, which are more looking for upstream compatibility and a future proof upgrade path, pure debian on raspi3 should be seen as a promising alternative as well.

bamarni, 2017-07-02:
Lately I've started to add a "pi64-update" command to update kernel / modules. There could be an "--upstream" flag to it. Feel free to contribute if you're familiar with how to build this kind of kernel.