bamarni/pi64 #13: Juniper VPN setup in RaspberryPi 3 with pi64 OS

jbustillosdt, 2017-06-20:
Hello friends. I have tried to setup a Juniper VPN in a RaspberryPi 3 using pi64 as operating system, because the necessity of execute Juniper binaries that are not available for ARM architecture with the official Raspbian image. Following some tutorials like this: http://blog.geeky.name/post/2016/03/29/HOWTO%3A-Ubuntu-Linux-64bit-Client-connect-to-Juniper-SSL-VPN-without-32bit-Java-(en) I tried first to do it with the icedtea-plugin to get the .jar files and searching a way with the offical Oracle Java Plugin, but only through icedtea is the option available for RaspberryPi and always fail during the Java API execution. After this, I tried with this great page of Arch Wiki following the content "Manual installation of msjnc": https://wiki.archlinux.org/index.php/Juniper_VPN but not success. Finally and currently I am trying these steps of a workmate:

1. Install the next packages as root:
# aptitude install stoken libc6:i386 zlib1g:i386 libgtk2-perl libwww-perl qemu libstdc++6:i386 libxext6:i386 libxrender1:i386 libxtst6:i386 libxi6:i386 build-essential cmake pcap-dev linux-headers-[latest]

2. Import the .sdtid file of our VPN with stoken as pi user:
$ stoken import --file file_name.sdtid

3. Create the Juniper network_connect directory as pi user:
$ mkdir -p ~/.juniper_networks/network_connect

4. Enter with Iceweasel to the URL of our customer login resource to obtain the ncLinuxApp.jar and download it.

5. Move ncLinuxApp.jar file to  ~/.juniper_networks/network_connect as pi user:
$ mv /home/pi/Downloads/ncLinuxApp.jar /home/pi/.juniper_networks/network_connect

6. Unzip the ncLinuxApp.jar file as pi user:
$ unzip ncLinuxApp.jar

7. Set the necessary privileges as root in the next files inside /home/pi/.juniper_networks/network_connect:
# chown root:root ncsvc
# chmod 6711 ncsvc
# chmod 744 ncdiag
# chmod +x getx509certificate.sh

8. Obtain the customer VPN certificate as pi user:
$ ./getx509certificate.sh customer.url.com file_name.cert

9. Execute the next command as root:
# echo 0 | tee /proc/sys/net/ipv6/conf/default/router_solicitations

10. And finally execute the connection to the VPN as pi user:
$ stoken --pin pin_number
84535943(random_generated_token_code_as_result)
$ ./ncsvc -h customer.url.com -u user_name -p 84535943(token_code) -r BlackBerry -f ./file_name.cert -U 'https://customer.url.com/dana-na/auth/url_9/login.cgi'

And when the execution finish, the next message appear in the shell prompt:
Connecting to IP_of_customer.url.com : 443
Unsupported ioctl: cmd=0x400454ca
pi@raspberrypi:~$

In the /home/pi/.juniper_networks/network_connect/ncsvc.log file I obtain the next output:

20170620100817.757984 ncsvc[p2503.t2503] ncsvc.info New ncsvc log level set to 3 (nccommon.cpp:75)
20170620100817.788892 ncsvc[p2503.t2503] sysdeps.info restoring DNS settings... (sysdeps.cpp:759)
20170620100817.791375 ncsvc[p2503.t2503] sysdeps.error rename /etc/jnpr-nc-resolv.conf => /etc/resolv.conf failed wirh error 2 (sysdeps.cpp:762)
20170620100817.793437 ncsvc[p2503.t2503] sysdeps.error rename /etc/jnpr-nc-hosts.bak => /etc/hosts failed wirh error 2 (sysdeps.cpp:766)
20170620100817.818755 ncsvc[p2503.t2503] ncsvc.info Connecting to IP_of_customer.url.com:443 (ncsvc.cpp:494)
20170620100818.174685 ncsvc[p2503.t2503] dsclient.info state: kStateSignin (dsclient.cpp:256)
20170620100818.175954 ncsvc[p2503.t2503] dsclient.info --> GET /dana-na/auth/url_9/login.cgi (authenticate.cpp:179)
20170620100818.204497 ncsvc[p2503.t2503] dsclient.info <-- 302 https://IP_of_customer.url.com/dana-na/auth/url_9/welcome.cgi?p=failed (authenticate.cpp:211)
20170620100818.205773 ncsvc[p2503.t2503] dsclient.info state: kStateWelcome (dsclient.cpp:264)
20170620100818.208268 ncsvc[p2503.t2503] dsclient.info --> GET /dana-na/auth/url_9/welcome.cgi?p=failed (authenticate.cpp:179)
20170620100818.357804 ncsvc[p2503.t2503] dsclient.info <-- 200  (authenticate.cpp:211)
20170620100818.361547 ncsvc[p2503.t2503] dsclient.info state: kStateLogin (dsclient.cpp:296)
20170620100818.363715 ncsvc[p2503.t2503] dsclient.info --> POST /dana-na/auth/url_9/login.cgi (authenticate.cpp:179)
20170620100822.657145 ncsvc[p2503.t2503] dsclient.info <-- 302 https://IP_of_customer.url.com/dana/home/starter0.cgi?check=yes (authenticate.cpp:211)
20170620100822.659096 ncsvc[p2503.t2503] dsclient.info --> GET /dana/home/starter0.cgi?check=yes (authenticate.cpp:179)
20170620100822.862424 ncsvc[p2503.t2503] dsclient.info <-- 200  (authenticate.cpp:211)
20170620100822.866624 ncsvc[p2503.t2503] authStateLogin.info starter0.cgi has asked for tz_offset parameter (authenticate.cpp:372)
20170620100822.871651 ncsvc[p2503.t2503] authStateLogin.info starter0.cgi has asked for clienttime parameter (authenticate.cpp:379)
20170620100822.875161 ncsvc[p2503.t2503] dsclient.info --> POST /dana/home/starter0.cgi?check=yes (authenticate.cpp:179)
20170620100823.50360 ncsvc[p2503.t2503] dsclient.info <-- 302 /dana/home/starter.cgi (authenticate.cpp:211)
20170620100823.51729 ncsvc[p2503.t2503] dsclient.info --> GET /dana/home/starter.cgi (authenticate.cpp:179)
20170620100823.233175 ncsvc[p2503.t2503] dsclient.info <-- 200  (authenticate.cpp:211)
20170620100823.236412 ncsvc[p2503.t2503] dsclient.info state: kStateAuthenticated (dsclient.cpp:376)
20170620100823.246444 ncsvc[p2503.t2503] IpcConn.info listening for IPC connections on port 4242 (ncipc.cpp:83)
20170620100823.266499 ncsvc[p2503.t2503] IpcConn.info unregistering the IPC acceptor IO handler (ncipc.cpp:125)
20170620100823.273019 ncsvc[p2503.t2503] IpcConn.info client opening connection to service (ncipc.cpp:319)
20170620100823.273788 ncsvc[p2503.t2503] session.info disconnectAll called (session.cpp:1648)
[... 30 lines cut ...]
20170620100824.85278 ncsvc[p2503.t2503] adapter.error Can not TUNSETIFF 38 (adapter.cpp:309)
20170620100824.85813 ncsvc[p2503.t2503] session.info onConnected calling disconnect for ive IP_of_customer.url.com (session.cpp:431)
20170620100824.86336 ncsvc[p2503.t2503] session.info disconnecting from ive IP_of_customer.url.com with reason 6 (session.cpp:506)
20170620100824.86686 ncsvc[p2503.t2503] adapter.info closing tun adapter FFFFFFFF (adapter.cpp:747)
20170620100824.88829 ncsvc[p2503.t2503] sysdeps.info restoring DNS settings... (sysdeps.cpp:759)
20170620100824.89360 ncsvc[p2503.t2503] sysdeps.error rename /etc/jnpr-nc-resolv.conf => /etc/resolv.conf failed wirh error 2 (sysdeps.cpp:762)
20170620100824.89770 ncsvc[p2503.t2503] sysdeps.error rename /etc/jnpr-nc-hosts.bak => /etc/hosts failed wirh error 2 (sysdeps.cpp:766)
20170620100824.96534 ncsvc[p2503.t2503] session.info disconnecting from ive IP_of_customer.url.com with reason 6 (session.cpp:506)
20170620100824.96963 ncsvc[p2503.t2503] adapter.info closing tun adapter FFFFFFFF (adapter.cpp:747)
20170620100824.97274 ncsvc[p2503.t2503] sysdeps.info restoring DNS settings... (sysdeps.cpp:759)
20170620100824.97554 ncsvc[p2503.t2503] sysdeps.error rename /etc/jnpr-nc-resolv.conf => /etc/resolv.conf failed wirh error 2 (sysdeps.cpp:762)
20170620100824.97812 ncsvc[p2503.t2503] sysdeps.error rename /etc/jnpr-nc-hosts.bak => /etc/hosts failed wirh error 2 (sysdeps.cpp:766)
20170620100824.98358 ncsvc[p2503.t2503] ncphandler.error NCP disconnect failed, error 107 (ncphandler.cpp:131)
20170620100824.98958 ncsvc[p2503.t2503] ncp.error ncpTearDown for IVE IP_of_customer.url.com (ncp.cpp:497)
20170620100824.99602 ncsvc[p2503.t2505] worker.error NCP worker has been requested to stop (ncp_dsssl.cpp:649)
20170620100824.100750 ncsvc[p2503.t2503] ncphandler.info disconnect done - tearing down (ncphandler.cpp:322)
20170620100824.104126 ncsvc[p2503.t2505] conn.info cleanup 0 (ncp.cpp:1418)
20170620100824.104877 ncsvc[p2503.t2505] writer.error thread exit (ncp.cpp:1848)
20170620100824.104945 ncsvc[p2503.t2503] ncphandler.info teardown done (ncphandler.cpp:340)
20170620100824.110486 ncsvc[p2503.t2503] ncp.error ncpCleanup for IVE IP_of_customer.url.com (ncp.cpp:618)
20170620100824.130999 ncsvc[p2503.t2503] session.info disconnected from ive IP_of_customer.url.com with reason 6 (session.cpp:569)
20170620100824.192462 ncsvc[p2503.t2503] ncui.info received onDisconnect with reason = 6 (ncsvc.cpp:628)
20170620100824.213766 ncsvc[p2503.t2503] IpcConn.error recv failed with errno 16 (ncipc.cpp:273)

Can someone explain me what is happening or help me to setup in a correct way this Juniper VPN in RaspberryPi please?

bamarni, 2017-06-23:
I'm not familiar with that software unfortunately, would it be possible for you to narrow down the problem? Is it that line : `Unsupported ioctl: cmd=0x400454ca`?

I can also see that you're installing kernel headers with apt, however as it's a custom kernel build they're probably not compatible, I'll add kernel headers in the next release (cf. #10).

bamarni, 2017-07-31:
@jbustillosdt : any news about this? Could you try to run `sudo modprobe tun` (cf. https://unix.stackexchange.com/questions/302623/opening-a-tun-device-from-qemu-x86-for-armv7-fails)?