bamarni/pi64 #15: Doesnt seem to connect to my WiFi network

ivankennethwang, 2017-07-06:
I've tried all of the settings with WPA2 but I can't seem to connect to my home wifi, nor to my phone hotspot.

Is there anything that I should do first?

bamarni, 2017-07-06:
How did you try to connect exactly? Have you tried with the `sudo pi64-config` helper tool?

ivankennethwang, 2017-07-06:
Yes, I tried both using the pi64-config and the one from the lower right system tray icon then manually setting the password of the wifi, then connect. When I went into the pi64-config and type i don't see a feedback though.

bamarni, 2017-07-06:
Could you paste here the output of the following commands?

    sudo ifconfig
    sudo route -n
    sudo systemctl status dhcpcd

bamarni, 2017-07-31:
@ivankennethwang : any news about this? Do you still have this issue?

ivankennethwang, 2017-07-31:
Sorry I havent had the chance to redo it again. I will check if reopen this issue if needed.

closed 2017-07-31 by ivankennethwang