bamarni/pi64 #16: Documentation on how to build and test?

abrightwell, 2017-07-17:
Would it be possible to add some documentation on how to build/test this project?

daxj, 2017-07-18:
Ooo, yes!  I strongly second abrightwell's notion: that would be a great help for numpties like me who enjoy using your superb images but could do with a simple guide on how to tweek it a little.  Especially helpful to me would be a note on how/when in the build sequence we could run "make menuconfig" or similar to setup extra kernel options we might like to play with.

bamarni, 2017-07-18:
I didn't focus on that yet but that would be nice indeed.

The simplest way is through docker : 

    git clone https://github.com/bamarni/pi64.git
    cd pi64
    docker run --privileged -v $PWD:/root/build -w /build ubuntu:17.04 make build/pi64-lite.zip

This should build the pi64 lite archive inside the build directory.

Otherwise, if you want to build on your host directly, ubuntu / debian is recommended (I don't know if / how it'd work with other distros), then just take a look at this file to make sure you have a similar setup : https://github.com/bamarni/pi64/blob/master/Dockerfile, and run the same make command.

bamarni, 2017-07-18:
As for the kernel config there is no interactive menu, it just expects the config to be in `make/kernel-config.txt`

Kosmatik, 2017-07-19:
pi64-update's main.go is missing os/exec in import. I also had to set a bunch of paths in /etc/sudoers and ~/.profile to get this thing to build on ubuntu 17.04, but finally got it!

daxj, 2017-07-19:
I've just tried building this on a totally fresh and squeaky-clean install of the latest 64bit lubuntu running in an Oracle VM VirtualBox.  It looked to get quite far but failed at:

```
<snip>
pi64-build -build-dir ./build -version lite
Creating build directory...
Creating image...
   Mapping partitions...
exit status 1
Makefile:6: recipe for target 'build/pi64-lite.zip' failed
make: *** [build/pi64-lite.zip] Error 1
```
I'm new to the world of Linux and Pi so please bear with me if I'm doing something utterly stoopid!

These are the steps I tried, from a fresh copy of lubuntu:

**Installed Docker**
```
sudo apt-get -y install curl git
curl -fsSL https://download.docker.com/linux/ubuntu/gpg | sudo apt-key add -
sudo add-apt-repository "deb [arch=amd64] https://download.docker.com/linux/ubuntu $(lsb_release -cs) stable"
sudo apt-get update
sudo apt-get -y install docker-ce
sudo usermod -aG docker ${USER}
su - ${USER}
```

**Setup Docker's DNS**
I found docker wasn't able to resolve web addresses, perhaps something to do with running in a VM(?).  Specifying the DNS in docker, however, seemed to allow it to work okay.
I made a new config file:
`sudo nano /etc/docker/daemon.json`
and entered the following
```
{	
  "dns": ["8.8.4.4", "8.8.8.8"]
}
```
Then restarted docker for it to take effect:
`sudo service docker restart`

**Cloned Bamarni's Superb Scripts**
```
git clone https://github.com/bamarni/pi64.git
cd pi64
```

**Made a couple of changes to Dockerfile**
- Added `libncurses5-dev `to the installed packages
- Added `TERM=xterm` to end of ENV line
- Added `&& go get -u golang.org/x/crypto/openpgp` to end of RUN line

**Made a couple of changes to make/Linux-src**
- Changed "mkdir -p build && cd build" to `mkdir -p build/boot-$version && cd build`
- Added `make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- menuconfig` after the line that looks like `make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- olddefconfig`

**Made one change to cmd/pi64-update/main.go**
Thanks to @Kosmatik for the heads-up here.
- Added "os/exec" to the import list

**Built the docker container**
docker build -t pi64 .

**Entered the container**
docker run -it pi64

**Launched the scripts**
make build/pi64-lite.zip version=lite

Not sure why I couldn't just run "make build/pi64-lite.zip" without the version variable but, when I tried without, it looked to be blank when Linux-src ran.

If anyone can help me get any further then that'd be marvellous!

bamarni, 2017-07-19:
@daxj : There are some capabilities needed in order to use loopback devices for example, I'm just using the `--privileged` flag when running docker run to be sure I have everything.

Also yeah there might be a few missing pieces in the master branch as it's still wip, if you use the latest tag that sould be more stable.

daxj, 2017-07-19:
Hazzah!  

Many thanks, bamarni, for your quick response.  I can confirm that I was able to compile an image to completion with the above steps (including the menuconfig step) once I corrected the docker run command as you suggested.  After changing the command to:
`docker run -it --privileged -v $PWD:/root/build -w /root/build pi64`
a pi64-lite.zip was successfully created, which I'll test tomorrow.  

Fingers crossed. :-)