bamarni/pi64 #17: Does the current version support google bbr?

cjl3230, 2017-07-25:
Does the current version support google bbr?
Google-bbr is useful, so I need the kernel 64bit with bbr run on the Pi3

bamarni, 2017-07-26:
I'm not familiar with google bbr at all, but I could find this thread on the raspberry pi kernel repo (https://github.com/raspberrypi/linux/issues/1784).

It seems like there needs to be a few kernel configs to enable it, feel free to submit a PR to allow this and I'll gladly merge.

cjl3230, 2017-07-27:
[image]

It default build as tcp_bbr.ko.  maybe change .config 
CONFIG_TCP_CONG_BBR=m

bamarni, 2017-07-31:
fixed by #18

closed 2017-07-31 by bamarni