bamarni/pi64 #19: USB does not seem to work after building

luishasmail, 2017-07-30:
Here is the problem I'm experiencing:

- I am running on Ubuntu 16:04
- I pull down the project and checkout the latest release tag (2017-05-23)
- I build the container and begin compiling
- Compilation completes like a charm
- I unzip the .zip file on my Mac, and write the resulting .img file using dd to an SD card
- When inserted into the Pi everything works as expected, my USB keyboard is completely unresponsive on all four ports.

Any suggestions on how to debug this?

bamarni, 2017-07-31:
Do you have the same problem with the zip release or is it only when building yourself from source?

If you set loglevel to 7 in cmdline.txt inside the boot partition maybe you'll see more info on the screen about what's going wrong.

luishasmail, 2017-07-31:
So the zip release works fine, but a compiled version of that tag does not.

I tried an older keyboard and things work fine with that keyboard, but I'm still not sure why the keyboard works with the zip file and not a compiled one

luishasmail, 2017-08-02:
I ran the PI through an upgrade and now both keyboards work. This isn't a big problem but I still find it interesting that there's a difference in the output. I won't have a chance to investigate this further for a while and since it isn't a huge problem I'm closing this.

closed 2017-08-02 by luishasmail