bamarni/pi64 #20: Cross-compiling apps

ronaaron, 2017-07-30:
I'm interested in cross-compiling on my main Linux box.  What I need is the appropriate sysroot to use (system-specific headers). 

What is the best way to get a set of such headers for cross-compiling?

bamarni, 2017-07-31:
Do you mean linux kernel headers? Starting from the latest release (2017-07-31), they should be available in /usr/include (cf. #10).

ronaaron, 2017-07-31:
body p { margin-bottom: 0cm; margin-top: 0pt; } 
  
  
    You mean in the running image?  Because I don't see them in the
      source tar.gz etc...
    
    
    On 07/31/17 05:38, Bilal Amarni wrote:
    
    
      Do you mean linux kernel headers? Starting from the latest
        release (2017-07-31), they should be available in /usr/include
        (cf. #10).
      —
      
        
          
          
        
        
      
    
    
    
      
        
          Ron Aaron | 
           CTO  Aaron High-Tech, Ltd |
          +1 425.296.0766 / +972 52.652.5543 |
          
            GnuPG Key: 91F92EB8

farfeduc, 2017-07-31:
Maybe you can apply the principles explained here : https://community.arm.com/iot/embedded/b/embedded-blog/posts/cross-compilation-for-arm and https://resin.io/blog/building-arm-containers-on-any-x86-machine-even-dockerhub/ to cross compile programs for aarch64 on x86 machines ?

You can find statically compiled qemu binaries here : https://github.com/multiarch/qemu-user-static

bamarni, 2017-08-02:
Yes in the running image, starting from 2017-07-31, or you can just grab them manually form here https://github.com/bamarni/pi64-kernel/releases (there is just the kernel / modules).

However I'm not sure if you need those or just cross compiling standard programs as shown by @farfeduc?