bamarni/pi64 #21: Update from 2017-05-23

sheychen290, 2017-07-31:
I have a "2017-05-23 lite" with my self hosted mail server.
It's work great, so it  is possible to install and run pi64-update on this release ?

bamarni, 2017-07-31:
`pi64-update` essentially just downloads the latest release from https://github.com/bamarni/pi64-kernel/releases and extracts it to the root filesystem. You could do it yourself manually for the first time, it'll give you an updated kernel and also the pi64-update binary.

There is also a file you'd need to create at `/boot/pi64.json` so pi64-update can work for later updates :
```
{"version":"2017-05-23","kernel-version":"2017-07-31"}
```

sheychen290, 2017-08-01:
Perfect thank

closed 2017-08-01 by sheychen290