bamarni/pi64 #22: Support other distributions

farfeduc, 2017-07-31:
Hi,

I like what you have build and I was wondering if you could merge all the logic related to debian into the *installDebian* function ? Because I think there is still some logic related to debian in the [finish.go](https://github.com/bamarni/pi64/blob/master/cmd/pi64-build/finish.go) file.

I would like to add a function *installAlpine* that would call https://github.com/alpinelinux/alpine-chroot-install with an answer file like described [here](https://wiki.alpinelinux.org/wiki/Alpine_setup_scripts). Do you think it might work ?

bamarni, 2017-08-01:
I think installing another distribution shouldn't cause much problems, pi64-config is indeed for debian but pi64-update should still work (it basically just updates kernel / modules).

Do you suggest to support this here or you would do your own project / scripts?

farfeduc, 2017-08-01:
I think it would be better to support this here. 
Having multiple arm64-bit distributions in the same place is much more interesting I think. 
What did you had in mind ? Do you think it would not be wise to have everything centered here ?

bamarni, 2017-08-01:
My main concern would be about how to document it and keep the doc understandable for both debian and alpine users. Otherwise I don't think it requires too much changes in the code.

In any case feel free to experiment and then we can see how to integrate it, I'm open to suggestions as well. I should probably turn this project into an organization to make it easier, and have different repositories for each distribution.

farfeduc, 2017-08-02:
What is the purpose of the documentation ? It may not be that difficult to keep it understandable.

I will experiment with what you already have to add alpine. I tried to build everything with your provided Dockerfile but I have an error in the kernel compilation. Do you have something else to do appart from executing `make` inside the docker container ?

bamarni, 2017-08-02:
For example most of the FAQ, and mentions of pi64-config are only for debian.

> Do you have something else to do appart from executing make inside the docker container ?
Are you running with the privileged flag? See https://github.com/bamarni/pi64/issues/16