bamarni/pi64 #24: Can pi64-update use local package?

Gaojianli, 2017-08-07:
Hello , I live in China , because some Chinese government's policy , the pi64-update can't download successful. Usually I use shadowsocksr proxy on my PC,but it doesn't work on pi yet. So the only way for me to update the kernel is download the package on PC, and then update manually.
I wonder if the pi64-update

Gaojianli, 2017-08-07:
can update my system by using a local package? Thanks

bamarni, 2017-08-07:
I'd suggest one of the following. Either you can find a proxy try with something like this : `sudo HTTP_PROXY=yourproxy:port pi64-update`? (Even though it's not your own proxy that's still safe because it will chech the pgp signature of the archive). Otherwise you can download the tar.gz manually from https://github.com/bamarni/pi64-kernel/releases, then extract it at the root of your filesystem. What do you think?

Gaojianli, 2017-08-08:
Well,I just exact the archive to the root system now ,but frankly speaking, it's very inconvenience and unsafe. By the way ,the shadosocksR use the socks5 proxy , and  most http proxies can be detected by the government. 

Interestingly, if I use the wget to download the archive, though it's slow(60-100K/S),I can still download it successful. When I use pi64-update , sometimes I get a "Couldn't verify signature : read tcp 192.168.10.145:58500->52.216.64.192:443: read: connection reset by peer",and sometimes I get a "network I/O error"

bamarni, 2017-08-08:
What about being able to run `sudo pi64-update 2017-XX-XX`, that will lookup for the archive in a cache directory otherwise download and install it? That would allow people to easily downgrade for example if they have a problem with an update, and in your case you could just drop the archive at the expected location.

Gaojianli, 2017-08-08:
That's a good idea. But I think being able to run `sudo pi64-update xxx.tar.gz` is better. As you say ,if people have a problem with a update ,then they may not  be accessible to the Internet. For a example , if you want to downgrade a package on a Linux PC , the usually method is downloading the proper deb/rpm to install it. What do you think about it.

Gaojianli, 2017-08-09:
OK , more is better. Thanks

bamarni, 2017-08-27:
I'd like to keep this open not to forget about this