bamarni/pi64 #26: libbcm_host.so missing

Gaojianli, 2017-08-24:
I have tried to install the RealVNC Server on pi64, which is pre-installed on most armhf distributions. As you can guess ,the RealVNC didn't provide any arm64 version. So I have to add the architecture armhf manually and install the armhf libc just like the ReadMe.md.
Later I installed it successfully by gdebi,I even can run vncpasswd. But if I try to execute the vncserver,I get this
 `# root @ raspberrypi in ~ [0:31:58]`
  `$ vncserver`
   `vncserver: error while loading shared libraries: libbcm_host.so: cannot open shared object file: No such file or directory`
How can I solve it? The results of google are for armhf, so none of them can help me.

Gaojianli, 2017-08-27:
Is there anyone can help me. Any advice is appreciated.

bamarni, 2017-08-27:
Right now it is not enabled for ARM64 (https://github.com/raspberrypi/userland/issues/391) but it doesn't matter for you as you're trying to run realvnc armhf.

It seems like you can find that shared library here for armhf : https://github.com/raspberrypi/firmware/tree/master/opt/vc/lib

closed 2017-10-20 by Gaojianli

piyushp132, 2020-02-19:
Hi Gaojianli,

Just wanted to know if you were able to make it work just by copying the shared library into /usr/lib?

p1r473, 2020-04-24:
@bamarni I am attempting to use a different application that requires libbcm_host.so on a TinkerBoard S.
I've downloaded the file, but putting it in /usr/lib doesn't work
How do I properly link or install this library?