bamarni/pi64 #27: how to enable SSH when first starting

YanShenChun, 2017-08-27:
Hi, bamarni

Thanks for the great share. I am using the pi64 desktop version but it seems the ssh is disabled by default.
I just want to know how I can enable SSH when starting the OS. For example, can we modify the /boot/config.txt to achieve this?

Any help is highly appreciated.

Gaojianli, 2017-08-27:
All you need  is to place an empty file named 'ssh' onto the boot (FAT) partition of your SD card. Or you can mount the ext4 on your PC and try to edit the rc.d. You'd better read the documents again:https://www.raspberrypi.org/documentation/remote-access/ssh/

bamarni, 2017-08-27:
> You'd better read the documents again:https://www.raspberrypi.org/documentation/remote-access/ssh/

Those instructions are specific to raspbian unfortunately.

SSH is not installed at all on pi64 desktop. So you'd have to be connected to the internet and install it yourself with `apt-get`. Is that an option? Or you specifically want it to be installed automatically?

Gaojianli, 2017-08-27:
>Those instructions are specific to raspbian unfortunately.

Oh ,I didn't notice that, sorry. I think the desktop version usually works with a monitor (TV), so you can download the openssh-server deb manually and install it with `dpkg -i.` If you don't have one ,you needn't to download a desktop version, which is much larger. Or you  use the lxde via VNC? If so, you can try to install the lxde on a lite version. Hope can be helpful.

YanShenChun, 2017-08-28:
@Gaojianli @bamarni  Thanks for the detailed answer. I just no keyboard in hand yesterday so I want to use SSH to connect PI.

closed 2017-08-28 by YanShenChun

ether-btc, 2018-02-20:
> Or you use the lxde via VNC? If so, you can try to install the lxde on a lite version. Hope can be helpful.
@Gaojianli 

I'd love to install lxde on lite version and also like to use realvnc, alas, I cannot get realvnc-server installed. It gets me the error
`unable to locate package real-vnc-server` and I am such a rookie, I cannot proceed.

I managed to install LXDE somwhow, but when I like to start it up with "startx", it says `/usr/lib/xorg/Xorg.wrap: Only console users are allowed to run the X server`