bamarni/pi64 #28: pi64-update progress bar

bamarni, 2017-08-27:
pi64-update downloads a ~50mb tarball, this can take some time so there should be a progress bar showing up.

olle2, 2017-09-14:
Hi bamarni, that sounds like a good idea. I tried the pi64-update yesterday, and it downloaded the update slowly.

Afterwards, I had to re-configure the config.txt file. It reverted to contain the dtparam=audio=on only and I had to disable overscan and force turbo again. Is this intentional in the pi64 update process?

Keep up the good work! Olle

bamarni, 2017-09-14:
Hi @olle2 ,

Thanks for reporting. Indeed the config.txt file shouldn't be part of the update tarball but in the distribution image only. I had this on the pipeline but didn't change it yet. I'll create an issue not to forget.

commit 9f002a53 (2017-12-02): [pi64-update] Add a progress bar (closes #28)
closed 2017-12-02 by bamarni