bamarni/pi64 #29: can't connect wifi

YanShenChun, 2017-08-28:
I tried to use 'sudo pi64-config' and choose the 'Configure wi-fi' setting. But I got a error as below:
'Couldn't scan for SSIDs'.

And back to the normal GUI intereface, I can see many WIFI connections in my current network.
But I can't connect my known WIFI.

Please help!

heduobacht, 2017-08-28:
I've got the same Problem. After playing a bit with the settings in WICD network Manager i got a wifi connection but it is not stable. After an hour or so the connection is lost and i have to play again with the wifi settings to get it connected again.

YanShenChun, 2017-08-28:
@heduobacht I just found WIFI work well after I reboot the PI.

heduobacht, 2017-08-28:
Did not work for me. After an hour or so the wifi disconnects itself without a reason.
Changed to Ethernet connection. This works great. For my project I can live without a wifi connection.

Gaojianli, 2017-08-28:
>Did not work for me. After an hour or so the wifi disconnects itself without a reason.
Changed to Ethernet connection. This works great. For my project I can live without a wifi connection.

Interestingly,i in my case the eth0 often disconnects itself, I have  to run `dhcpd eth0` to reconnect it via SSH on wifi.