bamarni/pi64 #31: pi64 cpu less neon support.

YanShenChun, 2017-08-28:
If I install the official 32bit OS, then the cpu feature output from 'cat /proc/cpuinfo' as below:
**_Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32_**

And If I install pi64, the output seem much less than above
**_Features	: fp asimd evtstrm crc32 cpuid_**

I just need the neon support. How to achieve this?

Any help is highly appreciated.

bamarni, 2017-09-11:
I'm not much familiar with this, isn't `asimd` the equivalent for arm64?

https://en.wikipedia.org/wiki/ARM_architecture#Advanced_SIMD_.28NEON.29

YanShenChun, 2017-09-13:
@bamarni  
Hi, bamarni
Yes, you are right. I had gotten this knowledge few days ago. Sorry to forget to share this point here.

closed 2017-09-13 by YanShenChun