bamarni/pi64 #33: Initial setup wifi issue

wscotgrey, 2017-09-06:
Hey There,

First, Thanks for this. 

Just a note for your setup. 

There is no nano,vi,or gedit that is installed with the full distribution that is referenced on the read.me.

You should mention that leafpad is available for text editing.

That is important because the Wicd Network Manager drops connections in the initial setup. 

To solve this, which took all day to figure out, you have to 
`sudo leafpad /etc/network/interfaces` and then follow the code on [here](https://learn.adafruit.com/adafruits-raspberry-pi-lesson-3-network-setup/setting-up-wifi-with-occidentalis).
I highly recommend saving the original file as `interfaces.bak` before changing anything.

This will let you input your wifi network id and password to hold the connection. 

Unfortunately, until you do this there is no accessing debian package files. 

Hope this helps
and thanks again!

bamarni, 2017-09-07:
What do you mean by `the Wicd Network Manager drops connections in the initial setup`? Doesn't it work if you configure it with their interface?

I also added wifi configuration to the `sudo pi64-config` command line tool.

wscotgrey, 2017-09-07:
in the GUI Wicd Network Manager, when you enter the details of your wifi setup, the application will connect to your wifi, then disconnect. Immediately. I had to figure out how to get the system to hold the connection. That is what wrote about. The key being that without internet access, or nano, or vi, or gedit, only leaped is available to edit config files. I saw nothing about a pi64-config in the readme. If it was there, sorry.

Thanks again for putting this together. Mad Props

S