bamarni/pi64 #35: Supporting intf. GPIO, I2C, SPI...?

Nic2012, 2017-09-10:
Hi,
this 64bit distro is working really fine, fast boot, wlan, eth, installing favourite apps no problem.
But is GPIO already included? Trying to install wiringPi and runs returns the following:
```
pi@pi64:~/wiringPi$ gpio readall
Oops: Unable to determine board revision from /proc/cpuinfo
 -> No "Hardware" line
 ->  You'd best google the error to find out why.
pi@pi64:~/wiringPi$ cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
```
Can I fix it or is that a matter of the Kernel?

Thank you for y effort to realize this project,
best reagards,
Nic

bamarni, 2017-09-11:
Hi,

I don't think this necessarily means there is something wrong with the kernel, it seems that on arm64 cpuinfo provides less information : https://github.com/raspberrypi/linux/issues/2110.

One could either submit a patch to the kernel so that arm64 also exposes this field or get wiringPI to have a detect mechanism that'd work for both archs (assuming they want to support arm64).

Nic2012, 2017-09-11:
You mean, without that patch we cannot use GPIO on pi64? Its matter by raspberrypi.org to provide a patch?
And whats about the other interfaces like: I2c, SPI are already working on pi64?

bamarni, 2017-09-11:
I wouldn't say you cannot use GPIO on pi64, it's just the command line tool from WiringPI which is not compatible, there are probably either library you can use for programming or other cli tools.

I haven't tested it yet so I cannot provide more pointers at the moment sorry.

Nic2012, 2017-09-12:
At least as root GPIO works in bash on low level:
```
echo "23" > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio23/direction
echo "1" > /sys/class/gpio/gpio23/value
...
```
and can be verified by LED.

Nic2012, 2017-10-08:
At least i2c is working fine, preconditions:
```
nano /etc/modules >>
i2c-bcm2708
i2c_dev
```
```
nano /boot/config.txt >>
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on
```
```
sudo apt-get update && sudo apt-get upgrade
sudo apt-get install i2c-tools
```
```
sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
Finally, a real-time clock DS3231 runs without problems.

Nic2012, 2017-10-22:
Well, I was successful to compile with modifications in original source code in RPi.GPIO-0.6.3 and Adafruit_Python_SSD1306, that finally this libs can be use in pi64. So in Python and RPi.GPIO i can now control gpios and with the Adafruit stuff I can access an oled display.

But this costs me a lot of research and trouble and waste of time. Mostly caused by this annoying "./proc/cpuinfo" issue. It would be much more easier when the kernel has the right patches.

Nic2012, 2017-10-23:
If nodejs v6 is installed, the only npm package that works for GPIO is so far https://www.npmjs.com/package/onoff. The others depending more or less on exact platform version.

closed 2017-10-23 by Nic2012