bamarni/pi64 #37: pi64 accelerated OpenGL or SDl2

vygr, 2017-09-28:
Hi there, great project BTW :)

I used PI64 as my host choice for my Linux Aarch64 compiler backend test hardware for my project. https://github.com/vygr/ChrysaLisp

I do have the port working now and the GUI running using SDL2, but performance is a little slow.

I know that there is hardware accelerated versions of OpenGL and SDl2 on the PI that people use but you have to enable the accelerated GL from the config tool. Is there a chance to add the option to pi64-config or another way to turn it on in PI64.

Best regards

Chris

bamarni, 2017-10-15:
Hi, I'm not much familiar with this, just looking at the current raspi-config code (https://github.com/RPi-Distro/raspi-config/blob/master/raspi-config#L1101), they do something like this : 

```
sudo apt-get update
sudo apt-get install libgl1-mesa-dri
echo "dtoverlay=vc4-kms-v3d" | sudo tee -a /boot/config.txt
# reboot
```

I don't have a raspberry pi at hand to test, could you confirm?

vygr, 2017-10-15:
After installing the mesa tools package I rebooted and this definitely seamed to run faster ! So it was probably some driver setting that came with the mesa tools.

Thanks

Chris

bamarni, 2017-10-15:
Didn't you also need the device tree overlay? I'll try to add this in `pi64-config`.

vygr, 2017-10-15:
I didn't appear to.....

vygr, 2017-10-15:
I don't have a monitor at home to connect my PI3 to :) So we're both without PI currently.

Chris

PeterLemon, 2017-10-26:
Hi guys,

I thought I'd give 64-bit ARM Linux on my R-pi3, & pi64 is easily the best I have found =D

I really want to get the HW OpenGL working, but following these instructions:
sudo apt-get update
sudo apt-get install libgl1-mesa-dri
echo "dtoverlay=vc4-kms-v3d" | sudo tee -a /boot/config.txt
# reboot

It still is using the CPU to drive 3D gfx...

I can tell this because when I run "glxinfo", a mesa info tool, the vendor ID does not match the broadcom HW accelerated driver information.

Also when running "glxgears" you can see the CPU usage goes very high, when it should remain low.
Running "glxgears -fullscreen" yields only 7fps on my 1920x1200 display, should be about 37fps with HW acceleration.

Also the game "Neverball" is a low unplayable fps.

I think bamarni is correct in thinking we need the full device tree overlay in place to get this to work.

I am happy to test for you guys, would love to get this to work in pi64.

Any help would be ace =D

vygr, 2024-04-12:
You can close this one now.

BTW much thanks for your efforts on this back in the day !

closed 2024-05-31 by vygr