bamarni/pi64 #38: logger doesnot write?

Nic2012, 2017-10-14:
I am bit confused, I did:
```
pi@pi64:~$ logger testmessage
pi@pi64:~$ sudo tail -1 /var/log/messages
tail: cannot open '/var/log/messages' for reading: No such file or directory
```
Is there any action needed before to enable logging systemwide?
Best regards,
Nic

bamarni, 2017-10-15:
Hi, seems like this file doesn't exist anymore, cf. https://unix.stackexchange.com/questions/84633/where-is-var-log-messages#answer-84634

Nic2012, 2017-10-15:
According to https://unix.stackexchange.com/questions/84633/where-is-var-log-messages#answer-84634
but i get the same to syslog, too :(
```
root@pi64:/home/pi# logger test
root@pi64:/home/pi# tail -1 /var/log/syslog
tail: cannot open '/var/log/syslog' for reading: No such file or directory
```

bamarni, 2017-10-15:
I don't have a setup to check right now, we should have a syslog daemon running right? (I think it's the `rsyslog` package by default). Could you also check with `journalctl -n`?

Nic2012, 2017-10-16:
I was expecting that such an infrastructure is already installed in y delivery, after installing rsyslog, I receive logs in /var/log/messages and an additional syslog was created, too.
Thx for y support.

closed 2017-10-16 by Nic2012