bamarni/pi64 #39: pi64-config didn't support 802.1x?

Gaojianli, 2017-10-20:
It seems that pi64-config only support WPA/WPA2-PSK. But in my university , the WLAN used 802.1x to encrypt. When i try to connect it ,the config only ask me for the password , not the username. I wonder if you can add the supporting of 802.1x

bamarni, 2017-10-24:
I don't have such a wifi access point to test against unfortunately, could you past the output `iwlist wlan0 scan` when you're at your university?

Gaojianli, 2017-11-06:
Here is what you need, and what I mostly connect is "BUPT-mobile":

**wlan0     Scan completed :
          Cell 01 - Address: A8:BD:27:19:27:42
                    Channel:6
                    Frequency:2.437 GHz (Channel 6)
                    Quality=19/70  Signal level=-91 dBm  
                    Encryption key:off
                    ESSID:"BUPT-guest"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 38ms ago
                    IE: Unknown: 000A425550542D6775657374
                    IE: Unknown: 010882840B162430486C
                    IE: Unknown: 030106
                    IE: Unknown: 2A0100
                    IE: Unknown: 32040C121860
                    IE: Unknown: 2D1AAD0917FFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D1606000000000000000000000000000000000000000000
                    IE: Unknown: 4A0E14000A002C01C800140005001900
                    IE: Unknown: 7F080500080000000040
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
          Cell 02 - Address: A8:BD:27:19:31:01
                    Channel:11
                    Frequency:2.462 GHz (Channel 11)
                    Quality=33/70  Signal level=-77 dBm  
                    Encryption key:off
                    ESSID:"BUPT-portal"
                    Bit Rates:9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s; 36 Mb/s
                              48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 38ms ago
                    IE: Unknown: 000B425550542D706F7274616C
                    IE: Unknown: 01079298243048606C
                    IE: Unknown: 03010B
                    IE: Unknown: 2A0100
                    IE: Unknown: 2D1AAD0917FFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D160B000400000000000000000000000000000000000000
                    IE: Unknown: 4A0E14000A002C01C800140005001900
                    IE: Unknown: 7F080500080000000040
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
          Cell 03 - Address: A8:BD:27:19:31:02
                    Channel:11
                    Frequency:2.462 GHz (Channel 11)
                    Quality=35/70  Signal level=-75 dBm  
                    Encryption key:off
                    ESSID:"BUPT-guest"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 38ms ago
                    IE: Unknown: 000A425550542D6775657374
                    IE: Unknown: 010882840B162430486C
                    IE: Unknown: 03010B
                    IE: Unknown: 2A0100
                    IE: Unknown: 32040C121860
                    IE: Unknown: 2D1AAD0917FFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D160B000400000000000000000000000000000000000000
                    IE: Unknown: 4A0E14000A002C01C800140005001900
                    IE: Unknown: 7F080500080000000040
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
          Cell 04 - Address: A8:BD:27:19:31:00
                    Channel:11
                    Frequency:2.462 GHz (Channel 11)
[... 118 lines cut ...]
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 38ms ago
                    IE: Unknown: 000B425550542D706F7274616C
                    IE: Unknown: 01079298243048606C
                    IE: Unknown: 03010B
                    IE: Unknown: 2A0100
                    IE: Unknown: 2D1AAD0917FFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D160B000000000000000000000000000000000000000000
                    IE: Unknown: 4A0E14000A002C01C800140005001900
                    IE: Unknown: 7F080500080000000040
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00**

By the way , I have connected it successfully by manually editing  wpa_supplicant.conf, maybe this can help you:

**ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
network={
	ssid="BUPT-mobile"
	key_mgmt=WPA-EAP
	eap=PEAP
	identity="{My id}"
	password="*****"
}**

Thanks for this project!