bamarni/pi64 #40: pi64 thermal regulation

phbillet, 2017-10-23:
Hereafter a small script for thermal regulation (usable with this electronic scheme [https://hackernoon.com/how-to-control-a-fan-to-cool-the-cpu-of-your-raspberrypi-3313b6e7f92c](https://hackernoon.com/how-to-control-a-fan-to-cool-the-cpu-of-your-raspberrypi-3313b6e7f92c)) : 
`#!/bin/bash`
`VALUE=45000`
`if [ ! -f "/sys/class/gpio/gpio18/value" ];then`
`chmod a+w /sys/class/gpio/export;`
`echo "18" > /sys/class/gpio/export;`
`chmod a+w /sys/class/gpio/gpio18/direction;`
`chmod a+w /sys/class/gpio/gpio18/value;`
`echo "out" > /sys/class/gpio/gpio18/direction;`
`fi`
`TEMP="cat /sys/class/thermal/thermal_zone0/temp"`
`if [ $TEMP -ge $VALUE ] `
`then echo "1" >/sys/class/gpio/gpio18/value`
`elif [ $TEMP -le $VALUE ] `
`then echo "0" >/sys/class/gpio/gpio18/value`
`fi`

Then using `sudo crontab` (with the entry  `* * * * *    /usr/bin/ctrl-fan.sh` ) it starts (or stops) the fans depending on the threshold (here 45°C).

As I got some problems with privileges, I had some chmod's : not really nice, but work. 
Do you know why ?

You should add the package zram (from ubuntu to your distribution)...

Your distribution works so nicely !

bamarni, 2017-10-25:
@phbillet : as which user your cron is running? If it's not root I think it's normal that you don't have write access.

phbillet, 2017-10-27:
When I tested the script, I used sudo, and the crontab is also root crontab (because I used sudo crontab -e). I used this trick, and it works fine but it is not really academic ;-)