bamarni/pi64 #41: Multiboot

xilopaint, 2017-11-03:
Is there any way to configure a multiboot of pi64, Raspbian and LibreELEC? I tried BerryBoot but [it seems not to be compatible with 64-bit](https://github.com/maxnet/berryboot/issues/353).

bamarni, 2017-11-03:
I'm not familiar with BerryBoot. But multiboot in general for pi64 would require manual tweaks, as at first boot it expands the root partition to use the full sd card.

In your case, what you'd want is to make multiple partitions, then having an init program that would display a list of the available partitions to choose from, and exec the init program of the chosen one. Probably this could be done with grub or similar, otherwise with a custom program.

I have never tried this, though I'd be curious to see how people would do this. Feel free to ask further questions here or share what you've found out.

xilopaint, 2017-11-03:
Unfortunately I have no clue how to achieve it. Pi64 is very nice for crypto mining but I miss my Kodi installation. 😞

bamarni, 2017-11-04:
Let's leave it open then, I'll try to look into it at some point and add a FAQ section about it.

xilopaint, 2017-11-04:
Yay! Thank you very much!

xilopaint, 2017-11-06:
I'm close to get the multiboot in NOOBS. At the moment it stuck on this:

[image]

Any idea?

xilopaint, 2017-11-06:
Hey @bamarni , NOOBS folks told me [this](https://github.com/raspberrypi/noobs/issues/379):

> @xilopaint - When you convert an OS for use with NOOBS, you need to remove such resizing options because they do not apply: NOOBS has already expanded the partition to the required size before the OS is installed. So the problem is not that it cannot expand the filesystem, but that it is trying to expand it at all. Find the boot script that causes this and delete it.

Where can I remove those resizing options?

bamarni, 2017-11-06:
@xilopaint : this is currently not possible, it is hardcoded in the init setup script. A way to avoid this could be to drop a file like `no-expand` in the boot partition, or have an interactive confirmation `expanding the root partition in X seconds, press [ESC] to cancel`. But it has to be implemented.

Could you explain how you set this up with noobs? What I was describing before is not optimal as this would make all OS running with the same kernel, so maybe noobs is a better option to document.

xilopaint, 2017-11-06:
Couldn't it be achieved commenting this:

```
fmt.Println("Expanding root partition...")
checkError(expandRootPartition())
```

and this:

```
func expandRootPartition() error {
	parted := exec.Command("/sbin/parted", "/dev/mmcblk0")

	// parted: "-1s" specifies exactly the last sector.
	parted.Stdin = strings.NewReader("resizepart 2 -1s\n")

	if out, err := parted.CombinedOutput(); err != nil {
		return fmt.Errorf("couldn't expand root partition : %s\n\n%s", err, out)
	}
	return runCommand("/sbin/resize2fs", "/dev/mmcblk0p2")
}
```

in `setup.go`?

Btw, I have set this up with [this tutorial for RetroPie](https://github.com/retropie/retropie-setup/wiki/convert-retropie-sd-card-image-to-noobs-image) and [this other one for Ubuntu Mate](https://github.com/raspberrypi/noobs/wiki/Adding-custom-OS-version-examples).

Btw you have to use `bsdtar` as mentioned in the second tutorial instead of `tar` mentioned in the first one.

xilopaint, 2017-11-07:
Hello @bamarni! Sorry for bugging again. Can you tell if your implementation should last long? If so I will have to search for another 64-bit distr for my crypto mining project.

bamarni, 2017-11-09:
@xilopaint : I won't be able to do this very soon, as a quick workaround if you have Go you can just comment that line `checkError(expandRootPartition())` and compile the `pi64-config` binary to `/usr/bin/pi64-config`.

xilopaint, 2017-11-12:
@bamarni 

I don't know anything about Go, but I have installed the version 1.9.2 and tried the following steps:

1) cloned the repo into `~/go/src/github.com/bamarni`
2) `$ cd ~/go/src/github.com/bamarni/pi64/cmd/pi64-config`
3) `$ go install`
4) `$ cd ~/go/bin`
5) `$ sudo cp pi64-config /media/user/pi640/usr/bin` # can't understand why the partition is called "pi640" rather than  "pi64"
6) After replacing the binary in the SD-card partition with the new compiled `pi64-config` and insert it in Raspberry Pi I get this kernel panic:

[image]

bamarni, 2017-11-13:
You have to cross-compile it for arm64, see for example how it's done here : https://github.com/bamarni/pi64/blob/master/make/linux#L43

xilopaint, 2017-11-13:
@bamarni 

Is it necessary even if I'm trying to compile in Debian?

Btw, I just tried some variations of the command an got errors in all of them. Could you tell me what I have to type exactly?

xilopaint, 2017-11-13:
Nevermind. I did it! Thank you so much!

closed 2020-06-28 by xilopaint