bamarni/pi64 #42: CPU temp

xilopaint, 2017-11-03:
How can I check the CPU temp in pi64? `vcgencmd` is not present.

bamarni, 2017-11-03:
What version are you using? vcgencmd should be present.

xilopaint, 2017-11-03:
Oh! It works with `sudo vcgencmd measure_temp`. I was missing that because in Raspbian we use `/opt/vc/bin/vcgencmd measure_temp`. Why is `sudo` necessary in pi64?

xilopaint, 2017-11-03:
Sorry, I found it! I was missing it because in Raspbian it's present in `/opt/vc/bin/vcgencmd measure_temp`.

closed 2017-11-03 by xilopaint

DanielSSilva, 2017-12-10:
Care to tell me the location @xilopaint pls?

xilopaint, 2017-12-10:
@DanielSSilva 

/usr/bin/vcgencmd

DanielSSilva, 2017-12-10:
using which, found it's on `/usr/bin/vcgencmd `
but when i run the command i get this

> pi@raspberrypi:~$ /usr/bin/vcgencmd measure_temp
> VCHI initialization failed

xilopaint, 2017-12-10:
Try using `sudo`.

DanielSSilva, 2017-12-10:
nice! that solved. Tks @xilopaint . Using it to mine and yet the temp is lower than with raspbian. Great jobe @bamarni !