bamarni/pi64 #43: About aarch64 binary compiling

leesera, 2017-11-03:
I have a raspberry pi3 model b and set that device with 64bit os.
the output of command 'uname -a' is like this.
`Linux raspberrypi 4.9.58-v8+ #1 SMP PREEMPT Sun Oct 29 16:25:42 JST 2017 aarch64 GNU/Linux`
I have a question about compilation at aarch64 kernel.

I want to make 64bit executable file with aarch64-linux-gnu-gcc-6.4.0
so when I try `aarch64-linux-gnu-gcc-6.4.0 test.c` , the output of that command is
`fatal error: sys/cdefs.h: No such file or directory`

I google it and I knew that I should install the needed lib like g++-multilib for arm64 version and for this, the arm64 version of apt repo should be set.

so I tried to add the debian arm64 apt repo. but I can not still install the suitable libraries for solving that compilation issues.
please help me for solving that problem!

bamarni, 2017-11-05:
Are you trying to compile from your raspberry or cross-compile? Eg. from amd64 to arm64?

leesera, 2017-11-28:
I compiled the 64bit kernel with cross-compile at my amd64 desktop

bao3, 2017-12-03:
@leesera  I suggest you to compile the binary in Docker ( you should docker build yourself  via Dockerfile in this repo)