bamarni/pi64 #46: First boot failure: couldn't run cmd dash.preinst

Qixuga, 2017-11-19:
Complete Error:

---------------------------------------------
couldn't run command [/var/lib/dpkg/info/dash.preinst install]:
dpkg-divert: error: parsing file ´/var/lib/dpkg/status´ near line 4663 package `libao4`:
 field name `Status*` must be followed by colon
---------------------------------------------

Nothing happens, also after a reboot.
I will check another sd-card to exclude a hardware-error. pi works fine with 32bit raspbian.

what can i do?
thank you!

Qixuga, 2017-11-19:
It seems that the sd-card may have a problem.
with another one it works fine!

Bug can be rejected!

bamarni, 2017-12-03:
Thanks for the info, I'll try not to have this step happening on first boot, see #50.

closed 2017-12-03 by bamarni