bamarni/pi64 #51: raspberry pi 3b can't connect wifi (wpa2 psk) ,but change conf can work

digitalgust, 2017-12-06:
using wicd Network Manger can't connect to my wifi hotpoint , it's wpa2 psk . 

#sudo iwlist wlan0 scan
          Cell 04 - Address: 34:36:3B:CA:07:8C
                    Channel:11
                    Frequency:2.462 GHz (Channel 11)
                    Quality=70/70  Signal level=-32 dBm  
                    Encryption key:on
                    ESSID:"36-1-601"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 5ms ago
                    IE: Unknown: 000833362D312D363031
                    IE: Unknown: 010882848B962430486C
                    IE: Unknown: 03010B
                    IE: Unknown: 050402030000
                    IE: Unknown: 2A0100
                    IE: Unknown: 32040C121860
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 2D1AAD4917FFFFFF0000000000000000000000000000000000000000
                    IE: Unknown: 3D160B000400000000000000000000000000000000000000
                    IE: Unknown: 7F080400000000000040
                    IE: Unknown: DD1B0017F20900010E4D6163426F6F6B50726F31312C31010103010000
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
in gui connecting , show on state bar , connecting, dhcp , all ok , but the last step done connecting failed .

then  i  found a method to solve the problem:
#sudo leafpad /etc/wpa_supplicant/wpa_supplicant.conf

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
#update_config=1
network={
   ssid="wifi_name"
   psk="wifi_password"
   priority=5
}

save it  may work or reboot work for me.

baronkwan, 2017-12-18:
Hi digitalgust,
Try this on the command line, `sudo iw reg set ES`, then connect to your wifi by insert the password again in the `sudo pi64-config`
Chances are your wifi channel is not detected by your pi.

shiftre, 2017-12-27:
this doesn't seem to work, I also manage to authenticate but it doesn't connect to the network.
i'm getting "scan error: (-11)" from the bcm network driver,

any idea how to solve it?

whitecider, 2018-03-19:
any progress? i am not able to connect to wifi. does seem to authenticate. not sure what digitalgust did im not an expert.

jmlcatarino, 2018-04-29:
Has anyone figured out this issue? I can't connect to any wifi network, no matter what I do.