bamarni/pi64 #52: Suspend, standby and hibernate

stevenwdv, 2017-12-06:
`rtcwake` and other suspend/standby/hibernate commands don't work because there are no modes available in `/sys/power/state`.
I read somewhere that this could be because the kernel wasn't compiled with `CONFIG_HOTPLUG_CPU` or `SUSPEND_SMP` on, but I'm not really sure what I'm talking about here.
Would it be possible to enable this feature?

erhanzeynel, 2018-01-08:
a Raspberry pi is a basic device on power management side as your hair dryer in hardware.
There is no switching devices to make it sleep, hibernate or powerdown. So compiling with power management flags will be pointless.

ThomasKaiser, 2018-03-27:
The ARM cores on any Raspberry Pi are just guests and the operating system running there is a secondary one. 

If you want to implement power management you have to do this on the main CPU (that's the VideoCore IV) using the primary operating system (ThreadX). But guess what... that's closed source and 100% proprietary so no way to do what you want unless you get hired by RPi Foundation or Trading.

Even to be able to access this stuff from the Linux kernel you would still need to access the proprietary VideoCore stuff since hardware is controlled there. The RPi is not an ARM SBC but something running something proprietary with some ARM cores added as co-processors.

If you want some insights here the main developer who tried to implement an open source replacement for ThreadX. She lost interest in the meantime for obvious reasons: https://irclog.whitequark.org/linux-sunxi/2017-04-02#19158376;