bamarni/pi64 #53: Is there an equivalent to rc.local?

DanielSSilva, 2017-12-10:
I'm fairly newbie into linux. I want to run a script upon starting up the pi. On raspbian i know we can use rc.local to achieve that , but I can't find that on this OS. What's the best way to achieve this? 
Thank you!

mohakshah, 2017-12-11:
You could use `@reboot` in your crontab. See [https://www.raspberrypi.org/documentation/linux/usage/cron.md](https://www.raspberrypi.org/documentation/linux/usage/cron.md)

DanielSSilva, 2017-12-11:
Totally forgot about crontab.... Why there's no such file as rc.local?

bamarni, 2017-12-11:
Seems like this has been removed by default on recent Debian (https://stackoverflow.com/questions/44797694/where-is-rc-local-in-debian-9-debian-stretch), you could also use systemd directly for this (https://askubuntu.com/questions/919054/how-do-i-run-a-single-command-at-startup-using-systemd).

DanielSSilva, 2017-12-11:
@bamarni  thank you for your answer. I've been using rc.local but I forgot that I was on previous version

closed 2018-01-15 by bamarni