bamarni/pi64 #57: Using /boot/setup

villermen, 2017-12-19:
I'm trying to create a RPi image from pi64 that simply has some packages, software and services added. To accomplish this I have added a setup file to the boot partition looking like this:

```bash
#!/bin/bash
apt-get update --yes
apt-get dist-upgrade --yes
apt-get install --yes

sudo -u pi git clone derr.git /home/pi/derr

cat <<EOF > /etc/systemd/system/derr.service
[Unit]
Description=Derr

[Service]
User=pi
ExecStart=/bin/bash -c "cd /home/pi/derr && git pull && ./derr.sh"

[Install]
WantedBy=multi-user.target
EOF

systemctl daemon-reload
systemctl enable derr.service

systemctl start derr.service
```

The setup file is processed in some way, as it is gone when the system is booted. It did not perform any of the actions listed above though. Is there any way to figure out what went wrong or why it is not running?

If I add and run the script manually afterwards it runs without failure, but I want to create a new image with the setup file without booting the system up first.

Any help is appreciated.

bamarni, 2018-01-03:
Hi @villermen, sorry for the delay, during the first boot the network setup is not set, systemd is also not running. So you cannot run git clone or these systemctl commands. You could just create a oneshot systemd service on the root partition directly.

villermen, 2018-01-06:
The one-shot service sounds interesting. If I were to create such a service, where on the boot partition should I place it in order for it to be picked up?

villermen, 2018-01-15:
In the end I wasn't even able to make the setup run:

```bash
#!/bin/bash

systemctl enable /boot/derr.service 
systemctl start derr.service
```

Even though manually running the commands worked flawlessly. I eventually resorted to just manually running the install script after I could log in through SSH for each of the RPi's.