bamarni/pi64 #69: Update and crush

netsafe, 2018-02-27:
After SD flashing it boots OK, `apt update` and `apt upgrade` - and here you go... no boot, no nothing. Power test is passed OK, PSU is able to give up to 3.1Amps, cable is good for 3Amps(checked)

ThomasKaiser, 2018-03-21:
Usually under-voltage or corrupted SD card. Either measure on pins 4 and 6 of the GPIO connector or connect a display and watch for the [yellow lightning bolt in the upper right corner](https://www.raspberrypi.org/forums/viewtopic.php?t=177477). If it's this you need to fix your underpowering situation.

netsafe, 2018-03-21:
5V 3A  runned stable, SD card was also OK, moved to debian image and it works. No yellow bolt was on the screen

joevwgti, 2018-04-05:
I can validate that apt-get update/upgrade pi64update breaks the install. Reboot just causes blank screen of nothing. I continue to watch for a newest release perhaps with updates packaged and tested/confirmed working.