bamarni/pi64 #70: what's the root password?

zhonghuabao1, 2018-03-05:
I need password when use Synaptic Package Manager

AurumNemo, 2018-03-24:
You can try to sudo passwd. I did this to get root  thought it isn't a good idea.

ThomasKaiser, 2018-03-27:
Why opening an 'issue' instead of RTFM or in this case simply reading the few sentences containing everything relevant: https://github.com/bamarni/pi64#getting-started

No password set, all you need is `sudo -s` for example (`man sudo`!)

jmlcatarino, 2018-04-29:
@ThomasKaiser how do you do that if the Synaptic P M is asking the password ? Sorry for my ignorance in this matter. I'm trying to install a program and I can't get around this.

PavelAnfimov, 2018-07-19:
> I got it around with AurumNemo trick.
@jmlcatarino  What the trick? I've go the same problem

jmlcatarino, 2018-07-19:
@PavelAnfimov I don't remember anymore, because I think I just gave up trying. I'm not that good with Linux. But I guess, at the time, I saw some kind of solution from a member with that name, so if you look for post or comments from that member, maybe you get to the answer. :(

bkauler, 2018-07-24:
I need it too. Developers, would you please tell us the root password.