bamarni/pi64 #73: Pi stuck after flashing .zip/img onto SD

Brewfasa, 2018-03-16:
Whenever i finish flashing the zip/img onto the Pi and boot it up i get stuck on this screen and its been over 20 minutes, isn't it working for me, or am i doing something wrong?

This is my screen https://i.gyazo.com/889d4342e04a7f4161069ae47efcc96f.jpg

romen, 2018-03-16:
That lightning icon on the top right corner means that your Raspberry Pi is not receiving the right voltage, which generally means it's underpowered: you should use a better power source!
For Raspberry Pi 3 it should provide at least 2.5A

michals91, 2018-03-16:
Have the same problem on my pi 3b+ (he probably also uses the same device). I'm using the official power supply. Probably it just lacks support for the new pi.

Brewfasa, 2018-03-17:
Yea i'm also using the pi 3b+ with the original charger, i guess every OS has to get updated to 3b+ since the normal 3 versions doesn't seem to work on the 3b+ yet.. I guess we're going to have to wait a couple weeks for that

ThomasKaiser, 2018-03-21:
> i guess every OS has to get updated to 3b+ 

No, it's only one OS that has to be updated. That's ThreadX which is the primary OS running on the primary CPU (dual core VideoCore IV). The ARM cores running the secondary operating system (eg. Linux) are only guests.

All the hardware initialization is done on the primary CPU cores by the primary OS. This one lives on the FAT(32) partition usually mounted as `/boot` and in official RPi OS images this will be updated as `raspberrypi-bootloader` package trying to overwrite the contents of the FAT32 partition. You could try to copy over the contents from most recent Raspbian light image or extract the stuff from the package directly: http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20180313-1_armhf.deb 

But for the new Ethernet chip also Linux support is needed (both driver and `.dtb`) so the best idea seems to be to check out the 4.14 branch and regenerate a new image then both containing latest ThreadX (AKA 'firmware') and RPi 4.14 kernel including driver and DT updates.

Wi-Fi then of course also needs new firmware. See the pending PR on `linux-4.14` branch.

ThomasKaiser, 2018-03-21:
Addendum: ThreadX is contained in these files:

    -rw-r--r--@  1 tk  staff     1494  9 Mär 19:28 LICENCE.broadcom
    -rw-r--r--@  1 tk  staff    52064 13 Mär 21:53 bootcode.bin
    -rw-r--r--@  1 tk  staff     6568 13 Mär 21:53 fixup.dat
    -rw-r--r--@  1 tk  staff     2592 13 Mär 21:53 fixup_cd.dat
    -rw-r--r--@  1 tk  staff     9721 13 Mär 21:53 fixup_db.dat
    -rw-r--r--@  1 tk  staff     9723 13 Mär 21:53 fixup_x.dat
    -rw-r--r--@  1 tk  staff  2823204 13 Mär 21:53 start.elf
    -rw-r--r--@  1 tk  staff   671556 13 Mär 21:53 start_cd.elf
    -rw-r--r--@  1 tk  staff  4966020 13 Mär 21:53 start_db.elf
    -rw-r--r--@  1 tk  staff  3910180 13 Mär 21:53 start_x.elf

If you use latest Raspbian Stretch image to copy files over please exclude the other files especially cmdline.txt [since incompatible](https://forum.openmediavault.org/index.php/Thread/18991-New-approach-for-Raspberry-Pi-OMV-images/?postID=170197#post170197).

Brewfasa, 2018-03-28:
Closing due to the same topic https://github.com/bamarni/pi64/issues/75

closed 2018-03-28 by Brewfasa