bamarni/pi64 #75: Bamarni64 not supported by Pi 3 b+

Brewfasa, 2018-03-17:
The current Bamarni 64 OS is not supported by the new Pi 3b+, when you try to load it after flashing it, it will hang on the rainbowscreen.

Would it be possible to update this so it gets supported?

ThomasKaiser, 2018-03-20:
You need at least

* new kernel with LAN7800 driver and correct device-tree contents
* new 'firmware' with support for the new hardware (this is the proprietary stuff on the `/boot` partition, the closed source RTOS controlling your Pi needs especially to know about the new PMIC and how to control it)

Solving the firmware part is somewhat easy (exchange the relevant stuff on the `/boot` partition with what you find on latest Raspbian Stretch image) but for kernel/driver support I would assume you need to do a little bit more :)

ThomasKaiser, 2018-03-20:
Just realized that everything should work as expected by building pi64 from source. Though I would switch to branch `linux-4.14` since most recent LTS branch and already supporting EEE (energy efficient ethernet) which can make a difference of over 400mW with the new USB Ethernet) https://github.com/bamarni/pi64/blob/cc3bfd245b8799efd23b8ea73c977ba487dd34d8/README.md

ThomasKaiser, 2018-03-20:
See also https://github.com/bamarni/pi64/pull/76 -- might work or not. Not tested since I lack the hardware.

Brewfasa, 2018-03-21:
Ah, that would be an easy solve. I will test it when i get home in about 11 hours from now and give an update if i managed to do so (I'm still quite new to Linux, so no promises, haha)

ThomasKaiser, 2018-03-21:
> Switching to 4.14 is easily done by just sending the command "sudo pi64-update" once its installed, right?

Don't think so. You need to check out branch `linux-4.14` before you let the image build in Docker. Check `git` documentation.

Brewfasa, 2018-03-21:
@ThomasKaiser do i have to compile the bamarni 64bit on a 64 bit os? I'm getting an error message when i try to build it on raspbian 32bit, this is what i get:

root@raspberrypi:/home/pi/Downloads/pi64-master# docker build -t pi64 .
Sending build context to Docker daemon 116.2 kB
Step 0 : FROM ubuntu:17.04
17.04: Pulling from library/ubuntu
1b3d7384fd40: Pull complete 
f9dcbbfaf1e9: Pull complete 
638a15e5eb12: Pull complete 
412eb9786954: Pull complete 
3c59bc4b5f14: Pull complete 
5a4ebf075c62: Pull complete 
Digest: sha256:92685bb08a498cf6229cd4a81a76763172020dab18a1a12b3b00fd008f9663db
Status: Downloaded newer image for ubuntu:17.04
 ---> 5a4ebf075c62
Step 1 : ENV GOPATH /go PATH /go/bin:/usr/lib/go-1.8/bin:$PATH
 ---> Running in cb366ce4d887
 ---> 2e736d57c419
Removing intermediate container cb366ce4d887
Step 2 : RUN apt-get update     && apt-get -y install         bc         build-essential         cmake         device-tree-compiler         gcc-aarch64-linux-gnu         g++-aarch64-linux-gnu         git         unzip         qemu-user-static         multistrap         zip         wget         dosfstools         kpartx         golang-1.8-go     && apt-get clean     && rm -rf /var/lib/apt/lists/*     && go get         github.com/aktau/github-release         github.com/cheggaaa/pb         golang.org/x/crypto/openpgp
 ---> Running in a1349699f50e
exec format error
[8] System error: exec format error
root@raspberrypi:/home/pi/Downloads/pi64-master# docker run -it --privileged -v $PWD:/root/pi64 -v /opt/vc:/opt/vc  -w /root/pi64 \
>     pi64 make build/pi64-desktop.img
Unable to find image 'pi64:latest' locally
Pulling repository docker.io/library/pi64
Error: image library/pi64:latest not found
root@raspberrypi:/home/pi/Downloads/pi64-master#

Brewfasa, 2018-03-21:
@ThomasKaiser I get exactly the same error message
https://pastebin.com/TV1xubYU

ThomasKaiser, 2018-03-21:
@Brewfasa you need to cross-compile on a PC. Either using a 'native' Ubuntu (I would assume 16.04 will already do it) or there using Docker pulling a more recent Ubuntu in.

The reason why: https://devsidestory.com/build-a-64-bit-kernel-for-your-raspberry-pi-3/

Brewfasa, 2018-03-22:
Alright, i got that point now, but for the:
> Solving the firmware part is somewhat easy (exchange the relevant stuff on the /boot partition with what you find on latest Raspbian Stretch image)

Is it possible for a noob like me to figure out which ones i have to copy and which ones i should keep? Like the noob i am, i tried copying the entire boot partition from raspbian to bamarni and it gives (kinda as expected) an error message during the boot.

I did notice that the raspbian boot files had something with kernal8 and bamarni kernal7, not sure if thats a problem

ThomasKaiser, 2018-03-26:
Without these commits from few hours ago building a 64-bit kernel with support for the new hardware wasn't working anyway: https://github.com/raspberrypi/linux/commit/74b4a1d8a7906120d69ef2312e503bc628590a8b

ArthurWebers, 2018-03-27:
I would also appreciate very much if pi64 wouldsupport the raspberry 3B+.

Brewfasa, 2018-03-27:
Maybe @ThomasKaiser could fork and update bamarni once the required up to date files are available.
He seems to be one of the few people who actually knows what he's talking about, unlike me, haha

ThomasKaiser, 2018-03-27:
Sorry, I already forked the project and commited the [outstanding PR](https://github.com/bamarni/pi64/pull/76) (untested). Now it's up to people who own the hardware to test. I don't have an RPi 3 B+ and I will never buy one since the hardware is sooooooo limited and slow compared to tons of other good SBC. And the under-voltage issues the majority of users is running into simply sucks.

The only reason I went through these annoying steps (mostly related to the proprietary and closed source crap on the `/boot` partition) is outlined here: https://forum.openmediavault.org/index.php/Thread/18991-New-approach-for-Raspberry-Pi-OMV-images/?postID=170817#post170817

Brewfasa, 2018-03-27:
Do we also have to do anything with https://github.com/raspberrypi/linux/commit/74b4a1d8a7906120d69ef2312e503bc628590a8b ?

Like i said, i'm a noob in this

ThomasKaiser, 2018-03-27:
> Do we also have to do anything with https://github.com/raspberrypi/linux/commit/74b4a1d8a7906120d69ef2312e503bc628590a8b ?

Nope. If you now build a new pi64 image the kernel gets checked out with those commits from yesterday so 3+ support should be there. Still missing is the wireless firmware but that's just copying a few files to a specific directory: https://github.com/bamarni/pi64/pull/76#issuecomment-376431384

Brewfasa, 2018-03-27:
@ThomasKaiser I tried to follow these steps on my ubuntu 64b in VMware. I cloned the linux 4.14 kernal, edited the finish.go file and went to the pi64 location.
git clone --depth=1 -b linux-4.14 https://github.com/bamarni/pi64 pi64

Edit finish.go to:
https://github.com/bamarni/pi64/pull/76/commits/62e3286e2df8bc5978793309bef8a22ec45751ae

cd pi64
docker build -t pi64 .

Then i run the docker build command and it starts downloading things from ubuntu and at the end i end up with the following error message:

https://i.imgur.com/3Oqk9zm.png

Reading package lists...
W: The repository 'http://security.ubuntu.com/ubuntu zesty-security Release' does not have a Release file.
W: The repository 'http://archive.ubuntu.com/ubuntu zesty Release' does not have a Release file.
W: The repository 'http://archive.ubuntu.com/ubuntu zesty-updates Release' does not have a Release file.
W: The repository 'http://archive.ubuntu.com/ubuntu zesty-backports Release' does not have a Release file.
E: Failed to fetch http://security.ubuntu.com/ubuntu/dists/zesty-security/universe/source/Sources  404  Not Found [IP: 91.189.88.152 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/zesty/universe/source/Sources  404  Not Found [IP: 91.189.88.162 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/zesty-updates/universe/source/Sources  404  Not Found [IP: 91.189.88.162 80]
E: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/zesty-backports/main/binary-amd64/Packages  404  Not Found [IP: 91.189.88.162 80]
E: Some index files failed to download. They have been ignored, or old ones used instead.
The command '/bin/sh -c apt-get update     && apt-get -y install         bc         build-essential         cmake         device-tree-compiler         gcc-aarch64-linux-gnu         g++-aarch64-linux-gnu         git         unzip         qemu-user-static         multistrap         zip         wget         dosfstools         kpartx         golang-1.8-go     && apt-get clean     && rm -rf /var/lib/apt/lists/*     && go get         github.com/aktau/github-release         github.com/cheggaaa/pb         golang.org/x/crypto/openpgp' returned a non-zero code: 100
root@ubuntu:/home/crazyhead/pi64#

ThomasKaiser, 2018-03-27:
No idea why you try this on an unsupported Ubuntu release. Zesty is EOL since months: https://wiki.ubuntu.com/Releases

ThomasKaiser, 2018-03-27:
I only use Ubuntu LTS releases (that's 16.04 at the moment, 18.04 is not ready yet)

ThomasKaiser, 2018-03-27:
Ah, now I get it. The Docker file still references Ubuntu 17.04. A simple change like this https://github.com/bamarni/pi64/pull/65 should be sufficient.

No idea why @bamarni didn't merge the PR but maybe he's busy with real life or something more important than RPi :)

bamarni, 2018-03-27:
Sorry guys indeed I haven't updated this repo for some time, there might be some stuff which don't work anymore, I plan to get back to this soon.

Brewfasa, 2018-03-27:
I downloaded, edited and build the kernal with docker but when i tried to create the desktop image using

docker run -it --privileged -v $PWD:/root/pi64 -v /opt/vc:/opt/vc  -w /root/pi64 \
    pi64 make build/pi64-desktop.img

I got this error:

  INSTALL include/video (3 files)
  INSTALL include/xen (4 files)
  INSTALL include/asm (35 files)
make[1]: Leaving directory '/root/pi64/build/linux-src'
++ make -s kernelversion
+ kernelversion=4.14.30-pi64+
+ hdr_dir=linux-headers-4.14.30-pi64+
+ echo ./.config
+ find . -name 'Makefile*' -o -name 'Kconfig*' -o -name '*.pl'
+ find arch/arm64/include include scripts -type f
+ find arch/arm64 -name module.lds -o -name Kbuild.platforms -o -name Platform
++ find arch/arm64 -name include -o -name scripts -type d
+ find arch/arm64/include -type f
+ grep -q '^CONFIG_STACK_VALIDATION=y' .config
+ find arch/arm64/include Module.symvers include scripts -type f
+ grep -q '^CONFIG_GCC_PLUGINS=y' .config
+ mkdir -p ../linux/usr/src/linux-headers-4.14.30-pi64+
+ ln -snf /usr/src/linux-headers-4.14.30-pi64+ ../linux/lib/modules/4.14.30-pi64+/build
+ unlink ../linux/lib/modules/4.14.30-pi64+/source
+ rsync -a --files-from=../hdr_files ./ ../linux/usr/src/linux-headers-4.14.30-pi64+
make/linux: line 37: rsync: command not found
Makefile:24: recipe for target 'build/linux' failed
make: *** [build/linux] Error 127
root@ubuntu:/home/crazyhead/pi64# 

Any idea what went wrong?

ThomasKaiser, 2018-03-27:
> rsync: command not found

This is missing: https://github.com/bamarni/pi64/pull/65/commits/8692e0aa63c2f5f4a03919d5b4d3fb453d555657

Seems you did not update the repo today? When you already did modifications to your local copy maybe best idea is to delete the pi64 directory and start over with a freshly checked out `linux-4.14` branch as [explained above](https://github.com/bamarni/pi64/issues/75#issuecomment-375063682)

Brewfasa, 2018-03-27:
It seems like the "master" repo is updated with the rsync command, but the kernal docker file hasn't, i guess thats the problem

Brewfasa, 2018-03-27:
Tried again to compile it with the rsync added to the kernel 4.14 Dockerfile. It went fine till where it started to create the .img and then it came with the following error:

pi64-build -build-dir ./build -version desktop
Creating build directory...
Creating image...
   Mapping partitions...
   Creating filesystems...
   Mounting filesystems...
Installing Debian...
   Running multistrap...
exit status 25
Makefile:15: recipe for target 'build/pi64-desktop.img' failed
make: *** [build/pi64-desktop.img] Error 1
root@ubuntu:/home/crazyhead/pi64# 

It doesn't really say what went wrong

Brewfasa, 2018-03-28:
Any idea what this could be @ThomasKaiser (just tell me if i'm asking to much of you all the time, but you seem like someone who's willing to help and know quite a lot, haha) & @bamarni

Brewfasa, 2018-03-28:
I found this code in install.go
Here you see the "Installing Debian", "Running multistrap" part, but before it hits the "Cleaning APT" it already printed the error message. So i guess its something in that part where the error lays.

https://i.imgur.com/T4sGbOo.png

bitvac, 2018-04-08:
I've ran into the same issue during the installing Debian stage of building the image.
FYI, I am building lite version.

Steps taken:

> git clone --depth=1 -b linux-4.14 https://github.com/bamarni/pi64 pi64
> cd pi64
> **Edit Dockerfile with:**
```
Line 1 "FROM ubuntu:17.10"
Line 22 "rsync \"
```
> docker build -t pi64 .
>docker run -it --privileged -v $PWD:/root/pi64 -v /opt/vc:/opt/vc  -w /root/pi64 \
    pi64 make build/pi64-lite.img

 

Resulting in the below after some time of compiling

> + cp -R firmware/boot/COPYING.linux firmware/boot/LICENCE.broadcom firmware/boot/bootcode.bin firmware/boot/fixup.dat firmware/boot/fixup_cd.dat firmware/boot/fixup_db.dat firmware/boot/fixup_x.dat firmware/boot/overlays firmware/boot/start.elf firmware/boot/start_cd.elf firmware/boot/start_db.elf firmware/boot/start_x.elf linux/boot
> touch build/linux # otherwise make will rebuild that target everytime (as build/linux-src gets altered by make/linux)
> pi64-build -build-dir ./build -version lite
> Creating build directory...
> Creating image...
>    Mapping partitions...
>    Creating filesystems...
>    Mounting filesystems...
> Installing Debian...
>    Running multistrap...
> exit status 25
> Makefile:12: recipe for target 'build/pi64-lite.img' failed
> make: *** [build/pi64-lite.img] Error 1

aalteirac, 2018-04-11:
Successfully built the img on ubuntu 14.04 but the PI (3B+) doesn't boot :-( Perhaps I missed something but I 've done: checkout of the linux-4.14, updated the docker file to use the ubuntu 17.10, added rsync.

Crazyhead90, 2018-04-12:
@aalteirac 
I just noticed this in the code: 
https://github.com/Crazyhead90/pi64/commit/522471c1eb98003bba548a270468b477c24fd6c3

It was still referring to the 3b boot files instead of the 3b plus boot files. I've updated it on my fork, but I'm not able to test it yet. If you have the time you could test it and let me know if it fixed it. But else I'll try it in a couple hours and let you know if it worked for me.

Crazyhead90, 2018-04-12:
@aalteirac how did you get it built on ubuntu 14.04?
Did you install it fresh and instantly built it?
I downloaded the 14.04, did some updates and i ran into the multistrap error again like in 16.04, 17.10 and 18.04..

aalteirac, 2018-04-12:
Actually my above statement was wrong, my ssh session died before the end of the build and I've seen the img created so I have assumed it was ok... I have the exact same error and will investigate tmrw if I can find the issue. Your PR can be super useful if we can get rid of this multistrap issue.

aalteirac, 2018-04-12:
oki, did few test, multistrap is working fine, if you run 14.04 edit /usr/sbin/multistrap and remove $forceyes on line 989.
Sound the problems is coming from go exec command. I have the same issue if I do multistrap manually, the next task using exec.command is failing too...

aalteirac, 2018-04-12:
here's the error : 
Using foreign architecture: arm64
multistrap building arm64 multistrap on 'amd64'
I: Setting /root/pi64/build/root-lite/lib64 -> /root/pi64/build/root-lite/lib symbolic link.
Getting package lists: APT_CONFIG=/tmp/multistrap.ufAbyc apt-get  -o Apt::Architecture=arm64 -o Dir::Etc::TrustedParts=/root/pi64/build/root-lite/etc/apt/trusted.gpg.d -o Dir::Etc::Trusted=/root/pi64/build/root-lite/etc/apt/trusted.gpg -o Apt::Get::AllowUnauthenticated=true -o Apt::Get::Download-Only=true -o Dir=/root/pi64/build/root-lite/ -o Dir::Etc=/root/pi64/build/root-lite/etc/apt/ -o Dir::Etc::Parts=/root/pi64/build/root-lite/etc/apt/apt.conf.d/ -o Dir::Etc::PreferencesParts=/root/pi64/build/root-lite/etc/apt/preferences.d/ -o APT::Default-Release='*' -o Dir::State=/root/pi64/build/root-lite/var/lib/apt/ -o Dir::State::Status=/root/pi64/build/root-lite/var/lib/dpkg/status -o Dir::Cache=/root/pi64/build/root-lite/var/cache/apt/ update
Ign:1 http://deb.debian.org/debian stretch InRelease
Get:2 http://deb.debian.org/debian stretch Release [118 kB]
Get:3 http://deb.debian.org/debian stretch Release.gpg [2434 B]
Ign:3 http://deb.debian.org/debian stretch Release.gpg
Reading package lists... Done
W: Download is performed unsandboxed as root as file '/root/pi64/build/root-lite/var/lib/apt/lists/partial/deb.debian.org_debian_dists_stretch_InRelease' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
W: GPG error: http://deb.debian.org/debian stretch Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 8B48AD6246925553 NO_PUBKEY 7638D0442B90D010 NO_PUBKEY EF0F382A1A7B6500
E: The repository 'http://deb.debian.org/debian stretch Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
apt update failed. Exit value: 100
exit status 25

Doesn't happen when running multistrap command manually...

Crazyhead90, 2018-04-12:
@aalteirac how do you get to see this error?

aalteirac, 2018-04-12:
with a redirect of the exec command output
cmd.Stdout = os.Stdout
cmd.Stderr = os.Stderr

Crazyhead90, 2018-04-12:
@aalteirac could you show me the code how you did it, i keep screwing it up trying to add it in there

aalteirac, 2018-04-12:
package multistrap

import (
	"os"
	"os/exec"
	"strings"
)

type Options struct {
	Arch       string
	Directory  string
	Suite      string
	Components []string
	Packages   []string
}

func Run(options Options) error {
	cmd := exec.Command("multistrap", "-a", options.Arch, "-d", options.Directory, "-f", "/dev/stdin")
	cmd.Stdout = os.Stdout
    cmd.Stderr = os.Stderr
	cmd.Stdin = strings.NewReader(`
[General]
noauth=true
unpack=true
allowrecommends=true
debootstrap=Debian
aptsources=Debian

[Debian]
source=http://deb.debian.org/debian/
keyring=debian-archive-keyring
components=` + strings.Join(options.Components, " ") + `
suite=` + options.Suite + `
packages=` + strings.Join(options.Packages, " "))

	return cmd.Run()
}

Crazyhead90, 2018-04-12:
Hmm it didnt work when i did it, but its working now after copying it, i guess i made a typo somewhere.
I'm running it now to see what error it tells me. Its easier to fix things when you see the changes

aalteirac, 2018-04-12:
I've screwed my env, I need to rebuild it. Not sure we are closed to solved the issue but at least progressing a bit. It sounds it could be easier to build without docker in that kind of situation. Fingers crossed bamarni will have look when he'll have time :-)

Crazyhead90, 2018-04-13:
`W: Download is performed unsandboxed as root as file '/root/pi64/build/root-lite/var/lib/apt/lists/partial/deb.debian.org_debian_dists_stretch_InRelease' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)`

This seems to be the issue, and i just cant get it to work. I tried putting user '_apt' in group 'sudo' and 'root' but than it still doens't have the permission to access the file.. 

I have no clue how i can get it to access that or to put the permission of that directory down. Maybe we could change the path to something else, but i'm also not sure how that works..

Maybe @ThomasKaiser could help us again after we managed to get a bit more detailed error message.

aalteirac, 2018-04-13:
After doing the multistrap manually and removed the instruction from install.go, exec.command (the next one is apt-get clean) continue to fail so I suspect something with GO and root permissions.  For apt-get clean it says something 'apt-get' command not found in $PATH... So every single error from install.go is related to weird user permissions (cp qemu-aarch64-static also fails)

Crazyhead90, 2018-04-13:
```
Creating build directory...
Creating image...
   Mapping partitions...
   Creating filesystems...
   Mounting filesystems...
Installing Debian...
   Running multistrap...
multistrap 2.2.9 using /dev/stdin
multistrap 2.2.9 using /dev/stdin
Using foreign architecture: arm64
multistrap building arm64 multistrap on 'amd64'
I: Setting /home/pi64/build/root-desktop/lib64 -> /home/pi64/build/root-desktop/lib symbolic link.
Getting package lists: APT_CONFIG=/tmp/multistrap.Y1ywnG apt-get  -o Apt::Architecture=arm64 -o Dir::Etc::TrustedParts=/home/pi64/build/root-desktop/etc/apt/trusted.gpg.d -o Dir::Etc::Trusted=/home/pi64/build/root-desktop/etc/apt/trusted.gpg -o Apt::Get::AllowUnauthenticated=true -o Apt::Get::Download-Only=true -o Dir=/home/pi64/build/root-desktop/ -o Dir::Etc=/home/pi64/build/root-desktop/etc/apt/ -o Dir::Etc::Parts=/home/pi64/build/root-desktop/etc/apt/apt.conf.d/ -o Dir::Etc::PreferencesParts=/home/pi64/build/root-desktop/etc/apt/preferences.d/ -o APT::Default-Release='*' -o Dir::State=/home/pi64/build/root-desktop/var/lib/apt/ -o Dir::State::Status=/home/pi64/build/root-desktop/var/lib/dpkg/status -o Dir::Cache=/home/pi64/build/root-desktop/var/cache/apt/ update
Ign:1 http://deb.debian.org/debian stretch InRelease
Get:2 http://deb.debian.org/debian stretch Release [118 kB]
Get:3 http://deb.debian.org/debian stretch Release.gpg [2434 B]
Ign:3 http://deb.debian.org/debian stretch Release.gpg
Reading package lists... Done 
W: GPG error: http://deb.debian.org/debian stretch Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 8B48AD6246925553 NO_PUBKEY 7638D0442B90D010 NO_PUBKEY EF0F382A1A7B6500
E: The repository 'http://deb.debian.org/debian stretch Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
apt update failed. Exit value: 100
exit status 25
Makefile:15: recipe for target 'build/pi64-desktop.img' failed
make: *** [build/pi64-desktop.img] Error 1
root@crazy-VirtualBox:/home/crazy/Downloads/pi64# 
```

Alright i managed to get past the permission error. I changed the ownership and permissions of /root/pi64/ (after i made the folder manually (mkdir /root/pi64)) with
```
sudo chown -Rv _apt:root /root/pi64/
sudo chmod -Rv 775 /root/pi64/
```
(Dont blindly change the permissions if you dont know the risks. I'm on a virual box, so i'm just trying everything)
Now i'm at the part where the no_pubkey error happens

Crazyhead90, 2018-04-17:
@aalteirac tried to compile the kernel and boot partition manually and uploaded it here:
https://github.com/Crazyhead90/pi64/releases/tag/2018-04-17

I'm not able to test it right now, the earliest would be like 5 hours from now. I think it should boot up, i'm just not sure if the wifi firmware works, but the only way to find out is to test it.

If anyone tested it, let me know if it worked or not

vixns, 2018-04-17:
@Crazyhead90 : tested, network / wifi / bluetooth ok, bravo !
 Could you plz build/upload a lite version ?

aalteirac, 2018-04-17:
I'll test, just takes time to d/l with my bandwidth :-) Also interested by the lite version ! Great job !

Crazyhead90, 2018-04-17:
Once i know that the desktop version works, i could do the same for the light yea

aalteirac, 2018-04-17:
I should be closed to compile it too, just need to find another box,14.04 seems not good to get it right. What distro did you use to build ?

Crazyhead90, 2018-04-17:
I used ubuntu 18.04 on VMware workstation player.

But i didnt compile it using his script, i used https://devsidestory.com/build-a-64-bit-kernel-for-your-raspberry-pi-3/

Crazyhead90, 2018-04-17:
I just tested the image in the 20 minutes time i had between getting home and having to leave again and it seems to work. The latest raspbian desktop version is now running on aarch64.
https://i.gyazo.com/e62356bdb7204b6634b1cab9991590ef.jpg

If you link me images of which you want me to try to convert to 64 bit, link them here. Else i could just try to do the raspbian lite into a 64bit version.

aalteirac, 2018-04-17:
Nice ! Will give it a try, tx for sharing !

aalteirac, 2018-04-17:
I have "firmware registration failed" and "Driver sdhost-bcm2035 is already registered", will try to reflash

aalteirac, 2018-04-17:
Reflashed, still same error but it load well and the wifi works

Crazyhead90, 2018-04-17:
Yea i have the same, not sure why it keeps checking to register it. But its registered so i guess its fine

Crazyhead90, 2018-04-17:
@vixns I'll make the lite version tomorrow my time (its 19:00 here and i only just got home, still got to shower etc)

aalteirac, 2018-04-17:
you deserve a shower :-) looking forward for the lite version but no rush, well done !

Crazyhead90, 2018-04-17:
Note that raspbian is a 32bit OS with a 64bit kernal now, if you guys want a full 64bit OS and kernel, i probably have to update Bamarni's release instead anyway.

Crazyhead90, 2018-04-17:
@aalteirac @vixns 
I got bored tonight, so i updated the bamarni lite with the new kernel.
The lite version is also in the https://github.com/Crazyhead90/pi64/releases/tag/2018-04-17.

I think it should boot, but now i'm not sure if the wifi or other things will work at all.
Let me know if any of you tried it.

aalteirac, 2018-04-17:
Started the d/l will tell you tmrw

Crazyhead90, 2018-04-17:
@aalteirac hold on, i just reuploaded the image, i forgot to change the wifi firmware. Restart the download and test the new one. (Same link etc, just the image got updated)

vixns, 2018-04-17:
@Crazyhead90 the lite image does not boot.

[32 more replies]