bamarni/pi64 #77: Running darknet-nnpack more slowly than pi32?

503612068, 2018-03-27:
I have updated pi3 from 32 to 64，and re-make darknet-nnpack[https://github.com/digitalbrain79/darknet-nnpack](url).But the test result is more slowly (16s) than old pi 32(8s). In fact ,pi64 running more faster than pi32 with CPU test command.
Darknet-nnpack needed NNPACK=1,ARM_NEON=1 to accelerate. I think,is it pi64 haven't supported or optimized？
Anyone who knows the problem?

ThomasKaiser, 2018-03-27:
> Anyone who knows the problem?

Under-voltage and you running frequency capped? Not already aware of Raspberry Pi problem number one? Shitty power circuitry?

If it's 8 vs. 16 seconds I would check whether you're running frequency capped at 600 MHz and ran with 1200 MHz before:

* https://github.com/bamarni/pi64/issues/4#issuecomment-291425512
* https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=208778
* https://github.com/raspberrypi/linux/issues/974#issuecomment-104701983
* https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=207888#p1285453

503612068, 2018-03-28:
I have checked my power voltage,it's not under-voltage and I try to set  frequency upto 1200MHz.It worked.But i found that my Raspberry Pi3 setup linux debian 32 no need to change frequency and worked fine. I still doubt pi64 isn't effect CPUs performance，yes？

ThomasKaiser, 2018-03-28:
@503612068 unfortunately you can't check for 'frequency capping' asking the kernel since `/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq` contains only fake values in situations where either frequency capping or throttling occurs. So you need to execute the following after you ran something heavy:

    perl -e "printf \"%19b\n\", $(vcgencmd get_throttled |  -f2 -d=)"

If this shows a `1` on the left you're affected. If this doesn't work simply try `sysbench` as shown in the first link above:

    sudo apt install sysbench
    sysbench --test=cpu --cpu-max-prime=10000 run --num-threads=4

6 seconds and you run under-volted, 3 seconds and at least the lightweight sysbench stuff is not able to trigger frequency capping.

> I have checked my power voltage,it's not under-voltage

How? Measured at pins 4 and 6 of the GPIO header or somewhere else?

ThomasKaiser, 2018-03-28:
You can also run in another terminal the following in parallel to get an idea about the real clockspeeds of the ARM cores:

    while true ; do
        vcgencmd measure_clock arm | awk -F"=" '{printf ("%0.0f\n",$2/1000000); }'
        sleep 1
    done

503612068, 2018-03-29:
3 seconds with command "sysbench --test=cpu --cpu-max-prime=10000 run --num-threads=4" in the pi64.

ThomasKaiser, 2018-03-29:
Well, the `sysbench` 'test' is the most unreliable one (since sysbench is such a light load). But I don't know whether `vcgencmd` is now available in pi64 or not. In case it is I strongly recommend as already suggested to try one of the two other methods outlined above. Either running your benchmark and then **afterwards** executing

    perl -e "printf \"%19b\n\", $(vcgencmd get_throttled | cut -f2 -d=)"

Or the other method [above](https://github.com/bamarni/pi64/issues/77#issuecomment-376820024).

ThomasKaiser, 2018-04-04:
BTW: Good news. With most recent kernel and 'firmware' Raspberry Pi problem N° 1 (under-voltage due to encouraging users to use crappy chargers and crappy 'power cables') has been somewhat addressed.

With latest kernel you can do an `dmesg | grep -i voltage` to get an idea whether you've been affected by under-voltage (and then frequency capping too I would assume).

https://github.com/raspberrypi/linux/issues/2367