bamarni/pi64 #79: pi64 lite v2017-07-31 missing text editor+wifi not working

gogu123456789, 2018-04-05:
Hello everyone!
I just installed pi64 on a raspberry pi 3. I tried running sudo pi64-config to configure my wifi. Although the machine sees my wifi network, when I'm asked to input my password, the form field seems to be locked. Any ideas how I can fix that?
I also tried to edit the /etc/network/interfaces file as I've seen here https://github.com/bamarni/pi64/issues/33. Unfortunately, leafpad doesn't exist! What text editor doespi64 lite version has?
Thanks!

ThomasKaiser, 2018-04-06:
Why do people follow outdated tutorials from hundred years ago? Putting Wi-Fi credentials into /etc/network/interfaces is insane.

I would do the following:

    rfkill unblock wifi
    reboot
    apt-get --no-install-recommends install network-manager
    nmcli dev wifi list
    nmtui-connect

`nmcli` will list wireless networks if everything works correctly and `nmtui-connect` will allow you to join the network (settings will the be stored as a secure profile -- you can change everything easily with `nmtui` later).

Important: No `wlan0` references in `/etc/network/interfaces` or below `/etc/network/interfaces.d`! Otherwise Network Manager won't touch the wireless interface and it can't work.

auto-joe, 2018-04-13:
@ThomasKaiser Where are you getting your guide from then? Enlighten us. Every other issue on here about wifi/wpa2 refers to pi64-config tool or /etc/network/interfaces, and they are less than 1 year old. This repository is not even updated monthly.

ThomasKaiser, 2018-04-14:
> Where are you getting your guide from then?

By not artificially limiting myself to the small Raspberry Pi monoculture. There are tons of better ARM boards available and around them you find tutorials that are not outdated as hell. If you really struggle setting up Wi-Fi on Debian Stretch (`pi64` is just an arm64 Debian Stretch distro with some additions to support the proprietary and closed source RPi stuff) then why not using the method that works and is easy even for beginners?!