bamarni/pi64 #81: benchmark

huichenggong, 2018-04-11:
I recently did some test and compared the Pi64 with raspbian. I'd like to share it hear.
[https://blog.csdn.net/weixin_38412284/article/details/79900850](url)
I wrote it in Chinese. Here is a brief introduction to the result.
[image]

In general, after the 64-bit system was used, the **integer operation** (sysbench) fully utilized the performance of the processor and was greatly improved. **Floating-point performance** (linpack) remained almost unchanged with a slight decrease. **Memory performance** (memtester) has decreased due to 64-bit reasons. 
In the application testing section, the GMX was tested. The overall strength of the 64-bit system was still good, with a certain degree of improvement.
Hope this will give you a brief insight into the performence of Pi64.

ThomasKaiser, 2018-04-14:
> Hope this will give you a brief insight into the performence of Pi64.

Unfortunately not. Every 'hardware performance' comparison including sysbench is just collecting numbers without meaning. Same with Linpack when done wrong (as here).

huichenggong, 2018-04-18:
sorry for the confusion numbers, I just want to compaire it with Raspbian.Can you please tell me what did I do wrong in the Linpack.

huichenggong, 2018-04-18:
thank you very much for pointing out my mistakes. I haven't done enough work on it.

closed 2023-10-12 by huichenggong