bamarni/pi64 #86: pi64-update installs kernel predating 3b+

satmandu, 2018-05-29:
Any chance of updating the kernels this tool installs to a newer version?

I see that the raspberry pi folks have a 4.16 tree now:

https://github.com/raspberrypi/linux/tree/rpi-4.16.y

@Crazyhead90 is there a workaround to reinstall the 4.14 kernel from your image build? (Or a good way to update to a newer kernel?)

Crazyhead90, 2018-05-29:
You could try following this guide:
https://devsidestory.com/build-a-64-bit-kernel-for-your-raspberry-pi-3/

Thats what i used to manually update the 3b to a 3b+ version

Instead of downloading the image in the guide, download my image and follow the kernel 4.8 part in the guide, but download 4.16 instead

satmandu, 2018-06-01:
It looked like 4.14 was getting more recent updates so I ended up doing something like this to update firmware & kernel:
```
apt update 
apt-get install -y bc build-essential gcc-aarch64-linux-gnu git unzip subversion
mkdir rpi ; cd rpi
# to get head, aka most actively developed current rpi kernel tree
git clone --depth=1  https://github.com/raspberrypi/linux.git
# for specific version use: git clone --depth=1 -b rpi-4.14.y https://github.com/raspberrypi/linux.git
# Now we get updated rpi firmware
svn export https://github.com/raspberrypi/firmware/trunk/boot
rm boot/config.txt
rm boot/kernel*
cd linux
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcmrpi3_defconfig
ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- make -j`nproc` deb-pkg
cp arch/arm64/boot/Image ../boot/kernel8.img
cp arch/arm64/boot/dts/broadcom/*.dtb ../boot/
cd ..
echo "kernel=kernel8.img" >> boot/config.txt
```
That should give a bunch of deb packages which can be installed on the target system as well as a boot folder which can be used to overwrite the boot folder on the target system. (This works for both 4.14 and 4.16, but I only tested 4.16 via ssh console so can't vouch for functionality like audio or video.)

closed 2018-06-01 by satmandu