bamarni/pi64 #92: Docker/Kubernetes support?

alexellis, 2018-08-28:
Hi I wondered whether you've verified if Docker/Kubernetes will work on this OS? A similar attempt exists from Hypriot where they add the required modules. There may be newer posts / sources - https://blog.hypriot.com/post/building-a-64bit-docker-os-for-rpi3/

HarshaVardhanJ, 2019-09-22:
Hey @alexellis! I'm currently running K3s on a 3-node cluster. All the Pis have the pi64 OS installed. No issues so far.

laimison, 2019-11-12:
Hi @HarshaVardhanJ that's good to hear. Anything specific to know? How did you build the cluster? Do you think kubespray could work as well? Here is Linux support list - https://github.com/kubernetes-sigs/kubespray#supported-linux-distributions