bamarni/pi64 #96: Windows 64-bit software on pi64 with wine?

goutham1220, 2019-03-02:
I've been having trouble trying install any sort of Windows software on pi64. I haven't even been able to download Wine properly. The failures for Wine may just be my fault but can anyone let me know how to install Windows 32-bit and 64-bit on pi64? If pi64 doesn't support that, are there any other 64-bit OSes that work on raspberry pi 3 b+ that I could use to install Windows software?

Gaojianli, 2019-03-02:
... I think you should learn more about computer ISA knowledge. The windows 64bit is working on X86_64, while pi64 is working on arm64. As the name showed, wine is not emulater, so most windows software can't work on wine. What you need is qemu, or you can try windows on arm64 instead

lucloner, 2019-04-27:
arm32/64 had wine, you could read https://wiki.winehq.org/ARM64
it could run win apps with kvm.
and it might support win arm apps.